Xinyuan-LilyGO/LilyGO-T-SIM7000G #1: New version of hardware discussion

lewisxhe, 2019-10-26:
In the previous version of SIM800L, many customers have put forward many good suggestions, so in the 7XXX version, the schematics are released first. Please give suggestions. We will adopt good comments in the new version of SIM7XXX series. Whether the chip is 7000G or 7600G, I will choose which one is the main chip in the following two replies.

Major changes regarding the new version:
1. Add TP4056 as the main power chip and as a solar charge management chip
2. Change the lithium battery to 18650 battery
3. Add SD card slot

Welcome to make suggestions, thanks
[SIM7000X.pdf](https://github.com/Xinyuan-LilyGO/T-SIM/files/3774314/SIM7000X.pdf)

lewisxhe, 2019-10-26:
Use SIM 7000G, like this

eueduardo3, 2019-10-26:
> Use SIM 7600, like this

these changes will be amazing!  Please do not forget that in all South America the compatible chip is SIM7600SA

nikil511, 2019-10-26:
- DTR connected on an ESP32 GPIO for controlling SIM7000 sleep/wake. 
When doing frequent network uploads (e.g. every minute) we do not want to power off and power on, register on network, turn GPRS. This wastes more energy than keeping the SIM7000 powered on but sleeping.  In sleep mode, the current consumption is reduced to the minimal level (<1ma) and
SIM7000 can still receive paging messages and SMS.

- Battery voltage monitor.
You could connect SIM7000 ADC pin directly on VBAT and we can use "AT+CADC" to get the battery when module is on. Or you can use a voltage divider on ESP32 S_VN like in T-CALL

- SOLAR_IN voltage monitor.
Again with a voltage divider on the ESP32 S_VN or S_VP we can measure solar panel voltage, which is very useful. 

- VBAT+ on a PIN (IO17/NC).
So we can connect an external battery or external fuel gauge, or voltage divider (in case you do not put one). I understand you plan to put a 18650 socket, which is good, but let's keep the option for an external battery.

- Expose MP5403 OUT3 on spare PIN (IO16/NC) and connect EN3 on an ESP32 GPIO.
It is very useful to have a spare voltage output for turning on/off peripherals, external sensors, etc. 

- Expose SIM7000 BOOT_CFG on a test pad, in case we want to flash the SIM7000 firmware.

- SIM7000 is better than SIM7600 for ESP32 in my opinion. 
**SIM7000** supports both **NBIoT and LTE-M**
SIM7600 supports only LTE-M
If you see the GSMA global network deployment, SIM7000 has double coverage, thus double market, than SIM7600. (SIM7000 should work in South America too, have you tried @eueduardo3 ?)
https://www.gsma.com/iot/deployment-map/
SIM7000 has very low power consumption on NBIoT mode, making it ideal for sensor applications. SIM7600 has more bandwidth/speed, so its better for a RPi HAT.

lewisxhe, 2019-10-31:
Thank you for your comments, we will listen to your suggestions. @nikil511

ClemensGruber, 2019-11-01:
please replace the battery plug with a JST PH 2-Pin as mentioned from https://github.com/Xinyuan-LilyGO/TTGO-T-Call/issues/13#

> Consider replacing your battery plug with standard "JST PH 2-Pin" that everybody are using.

I would vote for an option to switch off the SIM completely, we'd like to use the board in an solar powered and low power setup to send data only every x hours. So for a battery life over some month the SIM must be swtched off completely with a LOW pin, for a circuit see https://community.hiveeyes.org/t/low-cost-gsm-gprs-sim800-node/439/9

santiagobarros, 2019-11-07:
VERY USEFUL!  

> * Expose MP5403 OUT3 on spare PIN (IO16/NC) and connect EN3 on an ESP32 GPIO.
>   It is very useful to have a spare voltage output for turning on/off peripherals, external sensors, etc.

Electra99, 2019-11-11:
Have a SPI interface for a 2,4"-3.5" TFT display incl touch. (and not only for small 1" OLED displays with I2C) Please, have available GPIO pin for at least one SPI interface. Use SPI, HSPI or VSPI. Try to use VSPI (fastest) and avoid SPI. The TFT-displays have a common SPI interface for TFT and touch controller, but separate ChipSelect CS.
List of necessary GPIO pin for VSPI, HSPI and SPI to connect a TFT display
VSPI GPIO  23,19,18, 5 (fastest)
HSPI GPIO 13,12,14,15 
SPI    GPIO 7, 8, 6, 11

Electra99, 2019-11-12:
Good "PowerManagment" is always great to have when the device is powered by battery :-) 
E.q. A burgular alarm system with a IR-detector and sending SMS alarm message. 
Possibility to "Turn off" the power to SIMXXX module and turn off +5V power to the TFT/OLED-display.
Also possibility to connect chargeable 3.8V Li-ION battery or non-chargeable 3x1.5V standard battery e.q. AA or AAA.

lewisxhe, 2019-11-13:
Our hardware engineer changed the solution because the second power supply of the MP5403 power supply could not be turned on to normal 3.3v, so change the scheme to PDF, use the switch to select USB and battery.
[SIM7000X_V1.pdf](https://github.com/Xinyuan-LilyGO/T-SIM/files/3838893/SIM7000X_V1.pdf)

@nikil511 @santiagobarros @Electra99 @eueduardo3

Electra99, 2019-11-14:
The V1 board have :
PROS: 
1) Have VSPI on U6 connector (GPIO 18,19,23)  :-)
2) Have HSPI on U5 connector (GPIO 12,13,14)  :-)

Major CONS:
1) I looks in the schematics when the power failure occurs on the USB-connector it's NOT possible to have the battery as automatic power backup :-( 
Please, add P channel MOSFET for automatic power switching between the USB power and battery power. Control the MOSFET gate, for battery powered board, to the detect if USB/+5V i availible or not.

2) Missing the possiblity to turn off the power to SIM9000G for power saving and pure battery applications.
Please, add a PNP MOSFET to turn of the to the SIM9000G's VBAT pins 55, 56 and 57 and DVDD4V2.
Connect the MOSFET's gate to i GPIO pin.

Nice to have:
1) Add extra 4 hole (for a new connector ) to be used as a Piggy-back connector for a small and cheap ~1" OLED I2C display. Same as TTGO T-Display board. (pin1 GND, pin2 +3V3, pin3 SCL, pin4 SLA) or add the OLED direct on the board, because it's so cheap....

2) Add a extra 4 hole (for a new connector ) with 4xN channel MOSFET open collector output driver (1-6A) to be used for for e.q. external relays, 2 phase 6 wire Stepper motor NEMA17-23, PWM speed controlled 3-phase motor, BLDC-motor. Add a extra protection diod between each output and GND.

3) Add 1 or 2 small mechnical switches on the board and connect them to GPIO ports. These could be used for manual configuration e.q. to setup a alarm temperature and display on the OLED display. The ESP32 software could controlled  up to tripple functions for each switch by using click, double click and tripple click. This switch libarary exist on GITHUB.

nikil511, 2019-11-17:
Obviously is it vital to get power supply design right.  In general I agree with @Electra99 comments.

In fact, it would be nice to go for a more sophisticated power management design with a chip that has I2C and has fuel gauge / coulomb counter so we can measure mAh, as well as additional outputs that we can control over I2C. 

However a simple design is safer, robust and usually cheaper.
If this latest design has good efficiency on both BAT operation and SOLAR charging, then I am happy with this and I do not mind changing a switch/jumper upon field deployment.
I also do not mind that the external voltage monitoring will give a very rough idea of battery state and false picture during charging. 

A few N channel MOSFETs would be very useful and a better replacement for the missing "MP5403 OUT3"

Looking forward to testing an actual board (!)

lewisxhe, 2019-11-23:
[sim7000x.pdf](https://github.com/Xinyuan-LilyGO/T-SIM/files/3882098/sim7000x.pdf)
The new changes are as follows @nikil511 @Electra99

Electra99, 2019-11-27:
It look better ;-)  But, it would be great if the TTGO-board could detect and send a SMS alarm when the USB power has failed. Use ADC input on ESP32 and two resistors between GND and VBUS (for voltage division). Same solution as VBAT to IO35(ADC)

Gary-cat, 2019-12-12:
When would the new unit be avaliable to purchase. I have have just rolled out 10 of the current version.

bangonkali, 2019-12-12:
> It look better ;-) But, it would be great if the TTGO-board could detect and send a SMS alarm when the USB power has failed. Use ADC input on ESP32 and two resistors between GND and VBUS (for voltage division). Same solution as VBAT to IO35(ADC)

I support this. I have a client that requires notification if a node is potentially going to go down, ie busted/broken USB power bank or unplugged from AC. The smaller bat can take care of sending notification. More of a power supply redundancy in this case.

santiagobarros, 2019-12-13:
> However a simple design is safer, robust and usually cheaper.
> If this latest design has good efficiency on both BAT operation and SOLAR charging, then I am happy with this and I do not mind changing [...]

totally agree with this. Keeping it simple. 

Regarding the peripheral on/off option proposed using MP5403,  is there any alternative to switch the power or externar peripherals? Maybe WROVER GPIO controlled switch drawing current from usb 5v0 or from a duplicate of  ME6217-3.3V

ashley-schaeffer, 2019-12-22:
Since the SIM7000 seems to support GPS and NMEA output over either USB or the UART3_TX/UART3_RX pins it might be a good idea to also connect (or at least breakout) those pins?

This way the standard RX/TX pins can be used as dedicated modem pins for AT commands while simultaneously using UART3 for NMEA data. If those pins aren’t connected it will be next to impossible to use the modem and GPS module at the same time. (NMEA can be output over USB but reading it out over USB using the esp32 seems to be unnecessarily complicating things)

Disclaimer: I haven’t actually used the SIM7000 module so perhaps somebody who actually has can comment whether there is another workaround.

lewisxhe, 2019-12-23:
At present SIM7000 has passed the test and is preparing to be put on the shelves. It will be available for purchase soon.

Gary-cat, 2019-12-23:
This is awesome, I cant wait

Kind regards
Gary Jurgens
CTIO, Technical Architect.
Email: gary@overscope.co.za<mailto:gary@overscope.co.za>
Web: www.overscope.co.za<http://www.overscope.co.za>
Cell: +27  (0) 71 052 2778

lewisxhe, 2019-12-27:
Product is already available 
https://www.aliexpress.com/item/4000542688096.html

Gary-cat, 2019-12-27:
Awsome already contemplating ordering 10 in the new year. Does the Vin pin charge the battery in this version.

Gary-cat, 2019-12-31:
> Awsome already contemplating ordering 10 in the new year. Does the Vin pin charge the battery in this version.

Bump

lewisxhe, 2020-01-02:
@Gary-cat  As can be seen from the schematic, there is a VBAT pin, which is directly connected to the battery charging input

Gary-cat, 2020-01-02:
HI thanks

So it will power the whole board, I am just making sure before ordering 10.

Has Tiny GSM been shown to work on this board yet, with GPS Functionality?

@Gary-cat<https://github.com/Gary-cat> As can be seen from the schematic, there is a VBAT pin, which is directly connected to the battery charging input

lewisxhe, 2020-01-02:
It is possible to use TinyGSM, and GPS is fully working.
@Gary-cat 
[image]

Gary-cat, 2020-01-02:
Hi

Thank you, My apologies for all the questions, however, I live in South Africa and Buying volume (10 to start with) but potentially 200, in the long run, I need to be sure. And you seem like great developers. Does the 7000g support Active GPS Antenna ?.  I already have 2 of your units (T-Call) working in some of the harshest Coal Mines in South Africa for over 4 weeks now, and they work flawlessly despite the rough conditions. That’s how good I believe this product is.

Is Alibaba the only way to purchase these products from you?

[Image removed by sender. 123]<https://user-images.githubusercontent.com/22990954/71653876-afa71480-2d69-11ea-9061-735f053a7ae1.jpg>

lewisxhe, 2020-01-02:
@Gary-cat The SIM7000G board is powered by an active GPS antenna. Yes, only AliExpress is currently available, as if it is available on eBay.

Gary-cat, 2020-01-06:
Good Morning

Would you be so kind as to post a footprint of this modual or an EASYEDA Model, i need to design a PCB around it.

thank you

lewisxhe, 2020-01-06:
Do you need a dimensional drawing? @Gary-cat 

[image]

Gary-cat, 2020-01-06:
Hi

Thanks that would help

Also the underside for the battery so I can router the board so the battery fits through.

Thank you

Gary-cat, 2020-01-06:
Thanks you for the drawings, what is the length from the top of the board to the first pin, left and right, that the only dimension that is left out.

[Image removed by sender. F66KLZ(EZYI$X{@~_6FMU U]<https://user-images.githubusercontent.com/22990954/71797449-23f2f800-3089-11ea-9bb0-b8317e55ba8b.png>

lewisxhe, 2020-01-06:
Updated, thanks for reminding @Gary-cat

teezzan, 2020-01-13:
@Gary-cat Will the library be available for the public? And on what EDA will it be on? Thank you for your effort.

bangonkali, 2020-01-13:
Much appreciated if there is one available for EasyEDA.

teezzan, 2020-01-13:
Eagle will be cool too...

hardmax, 2020-01-13:
Hi 

Sorry, maybe is a stupid question but, Why IO21 IO22 IO32 and IO33 are pulled up?

Thanks for yu answer

lewisxhe, 2020-01-14:
@hardmax Thank you for reminding, 32, 33 This should be redundant and can be removed, 21, 22, in fact it can be retained, because the design can use the general I2C module combination

ClemensGruber, 2020-01-15:
Has someone tested the deep sleep current of the system already? SIM7000 seems to be promising with 9 uA in Power Saving Mode (PSM). AT command ”AT+CPSMS=1” can enable the PSM mode.

jbrepogmailcom, 2020-01-21:
> It look better ;-) But, it would be great if the TTGO-board could detect and send a SMS alarm when the USB power has failed. Use ADC input on ESP32 and two resistors between GND and VBUS (for voltage division). Same solution as VBAT to IO35(ADC)

Hello - would the voltage divider cause a constant current drain? If voltage divider is really needed, then I would connect its ground to GPIO pin pulled up, so no current is drained. Then when I want to measure voltage, I would set that GPIO LOW, measure voltage and then set it back up or leave floating. What do you think?

ahmadfarisfs, 2020-01-22:
Hello, what a great board ! But How do I measure battery voltage with this board ?

nikil511, 2020-01-22:
GPIO35 should have a voltage devider on vbat according to schematics. My
(beta) hardware was missing the upper resistor, so in case you read zero,
you need to add a 100k ohm between vbat and pin 35

ahmadfarisfs, 2020-01-22:
I cannot find any schematic in this repo, where do you get the schematic?

roberto1984, 2020-01-22:
Dears, I have a question for this board and also for the 2g version with sim800l..
If I connect to the internet with data plan of the sim card,then is possible to set the board to act as an access point and allow other device connecting to it to send data on internet through the previous established data sim connection?
Thanks in advance

lewisxhe, 2020-01-28:
@roberto1984 Are you talking about data forwarding?
esp32-> sim800 / sim7000-> internet?
This is ok and needs to be handled on the code yourself

lewisxhe, 2020-01-28:
We will check @nikil511

lewisxhe, 2020-01-28:
This issue is too long, I will close it, if there is a problem, please re-open a theme

closed 2020-01-28 by lewisxhe

roberto1984, 2020-01-28:
@lewisxhe no I'm talking about use a single board,for instance the one with sim800l,using sim to establish an internet connection and once established let other devices send data on internet connecting via wify to the board using the previous established connection with SIM card.

lewisxhe, 2020-01-28:
You said that sim800 establishes a connection with the Internet and sends data to the sim800 single board through wifi? @roberto1984

nikil511, 2020-01-28:
Esp32 as an internet router ( wifi hotspot with gsm backhaul ) will be very
slow, there are various limitations, and this is not Arduino friendly (or
possible) requires RTOS, NAT ip,port managment and more.

A project that could evolve that way does exist but I think there is a lot
of extra work that needs to be done with no obvious value.

https://www.esp32.com/viewtopic.php?t=1496

karioja, 2020-02-13:
I was pleased to find the DTR is connected. However, why leave RI unconnected? Connect RI to an interrupt capable pin on the ESP32. This way you would be able to put both the SIM7000 into sleep and ESP32 into deep sleep and have the LTE module wake up ESP32 when there's any activity (incoming SMS for example). I've been searching for a good SIM7x00 module a long time. Most of the modules on ebay and aliexpress are crap. They just connect VIN, GND, RX and TX. No DTR. No RI. This module came so close.

lewisxhe, 2020-02-14:
Yes, this is indeed not reserved and will be added in the next revision @karioja

hardmax, 2020-02-17:
Hi, 
I try with this hardware and Its great. But with the battery Holder in back, Its dificult put conectors or maybe a Hat to connect anothers thinks, like sensors or anything else. Another thing its size, Its too large and is dificult find a water probe case. Maybe if you remove the battery Holder and replace it with a conector, and change shape of PCB for a square (smaller) will be great...

Gary-cat, 2020-02-18:
Hi

I agree 100% it does make it difficult but not impossible to use. I just double stack my female headers. It rises it enough.

What would be nice is an Eagle / Easy EDA Footprint of this so we can make PCBS and hats for this.

Also what pins are the strapping pins on the ESP32 on this model. ?

Kind regards
Gary

thomas-rode, 2020-04-02:
Hi,
on GPIO35 I measure withn multimeter 2,05 V (its Ubatt/2), but with the example code, I read 0V (the solar voltage works fine!). I belive, this is no problem with the 100k pullup, must be some other problem ... any idea?

MalharJojare, 2021-05-12:
Anyone worked on SIM7600 for data transmission and reception using Arduino/STM32/ ESP8266..............