Xinyuan-LilyGO/LilyGO-T-SIM7000G #3: Deep sleep power consumption

ClemensGruber, 2020-02-03:
Has someone tested the power consomption in deep sleep already. I found on 

http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1246
> Sleep Current: 1.2mA

This would be more than the previous model, the TTGO T-Com needs. SIM7000 should be much less power hungry in deep sleep. And I think there are some mechanisms to switch off power compeletely for the SIM7000? [At least it was discussed. ](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues?q=is%3Aissue+is%3Aclosed)

nikil511, 2020-02-05:
yes, I have tested it and was near ~1mA
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/4

ROSW6341, 2020-02-08:
Was the current measured with an active GPS antenna connected?

ROSW6341, 2020-02-08:
I'm not sure this pertains to the issue of unexplained deep sleep power consumption, but are the signals to the source and drain of Q7 (V_BAT & VVBAT respectively) by chance reversed?

eiriksels, 2020-02-10:
Hi. 

I have done some initial testing and I think it is draining the battery much faster than I expected. I first tried to just put the ESP in deep sleep, but for this board it seems that the SIM7000 stays alive(Unlike the T-Call board)? I use the TinyGSM Library.

I use this to enable sleep: 
esp_sleep_enable_timer_wakeup(Time);
esp_deep_sleep_start();

I therefore included a modem turnoff command( modem.poweroff();) in addition to the sleep command. 

Could you give me any advise on how you are putting the board to sleep between sending values?

nikil511, 2020-02-10:
https://github.com/nikil511/LilyGO-T-SIM7000G/tree/master/examples/Platformio_Arduino_Thingsboard

eiriksels, 2020-02-11:
@nikil511 Thanks a lot. It turns out that it was my network settings that caused the battery usage. I was actually on 2G and not NB IoT as wanted. I will create a separate issue for this setup because I have not been able to use the AT debug sketch with this board to manually set the operator setup.

eiriksels, 2020-02-12:
@nikil511 Did you disable or remove the blue onboard power-indicator led to reach the 1mA consumption?

nikil511, 2020-02-19:
@eiriksels nope...but..there is no power indicator led on my version of the hardware. :-/
(maybe something is different in the final version, I have beta hardware)

kapitan-u, 2020-02-27:
Improvement idea: 100k/100k divider for measuring battery power is causing excessive power draw.
please consider P-FET solution described here
https://fettricks.blogspot.com/2014/01/reducing-voltage-divider-load-to-extend.html
for future versions

ROSW6341, 2020-02-27:
Improvement idea #2:  The GPS antenna circuit applies supplies power for active antennas.  Unfortunately there is no means to switch the power off when the SIM7000 is shut down.  An active antenna will continue to draw about 10mA even when the ESP32 and SIM7000 are shut off.  A P-channel FET could be used as a switch with an N-channel FET's drain tied to its gate to act as a level converter (the 4+ volt battery supply is too high for either the ESP32 or the SIM7000).  The N-FET's gate would be tied to the SIM7000 1.8v V-EXT.  Pulling the N-FET's source low would enable the switch.

ROSW6341, 2020-02-27:
In response the kapitan-u's post.  The 100K/100K divider would draw about 20 uA assuming a battery voltage of 4 volts.  A reasonable assumption for the capacity of a 18650 battery is in the area of 3000 mAH.  So it would take about 150,000 hrs for the resistor divider to deplete the battery.  An active antenna with the current feed circuit will draw 500 times as much.

nikil511, 2020-03-11:
@ROSW6341 
I tested sleep consumption with GPA antenna connected and its 36ma (!)
Tried AT+CGNSPWR=0 same thing
AT+CVAUXS=0 gives error

So, yes, I confirm there is a serious sleep current consumption problem!

Elmi77, 2020-05-28:
So can anybody recommend a passive GPS antenna that works with this board?

yevgenb, 2020-06-08:
Can we power off SIM7000 or GPS only completely to reduce consumption?

ROSW6341, 2020-06-08:
I retested the current draw in deep sleep mode with an active GPS antenna on v1 of the board.  This and the previous test above were done using an in-line USB voltage/current meter.  The previous reading of 36ma was in error because the USB com port was open being monitored by a terminal emulator.  The new reading of 16ma was taken with the same setup but without opening the com port (no terminal emulator).  That value is more in-line with what I would expect for the current drawn by the green 'battery charged' LED and an active GPS antenna.

ROSW6341, 2020-06-08:
> Can we power off SIM7000 or GPS only completely to reduce consumption?

Both can be powered off.  See the SIM7000 Series AT Command manual.  The SIM7000 Hardware Design documentation describes an additional way to power off the SIM using the PWRKEY.

ROSW6341, 2020-06-08:
So can anybody recommend a passive GPS antenna that works with this board?

The next version of the board (now available on AliExpress) has a means to control the power to an active GPS antenna.

Elmi77, 2020-06-14:
> So can anybody recommend a passive GPS antenna that works with this board?
> 
> The next version of the board (now available on AliExpress) has a means to control the power to an active GPS antenna.

Sounds good! How can one recognise if the correct (new) version is offered? I think there are a lot of older boards still floating around...

Mr-HaleYa, 2020-06-16:
@ClemensGruber The new version has better power consumption due to the GPS antenna now being able to be completely disabled, If your question has been answered please close your issue

ClemensGruber, 2020-06-16:
I do not use the GPS function so I'm interested in the lowest possible deep sleep power consumption. We have a long term measurement project with solar power support. I found values between 1 mA in sleep mode and 7 uA in power off mode for the SIM7000x series, see https://www.simcom.com/product/SIM7000X.html so I wonder why there is still a sleep current of about 750uA on the web page. Has someone tried do power off the SIM7000 and measured the depp sleep current of the new version?

lewisxhe, 2020-06-18:
The new version has been updated, the issue will be closed, if there are other issues, please open the issue

closed 2020-06-18 by lewisxhe

ClemensGruber, 2020-06-18:
@lewisxhe you are writing at https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/11#issuecomment-626593667 

> About the latest version, the test has been completed, and the final test sleep is about 300uA.

But on the product page https://www.aliexpress.com/item/4000542688096.html you can read Sleep Current:Without GPS antenna, sleep current is about 750uA

May you share your deep sleep testing code? And do you know why there is a far higher amount on the product page?

lewisxhe, 2020-06-18:
I have told them to let them delete, this sleep current is only the old version, the new version of the sample code when the test is completed, go to sleep about 300uA, this is my actual test!
@ClemensGruber

javipelopi, 2021-03-10:
@lewisxhe hi!

I would also like you to upload the code you used to get to 300uA and to know if you desoldered something to get to that consumption.

Thanks in advance and amazing job you are doing here!

ishdemon, 2022-07-21:
i got rid of the antenna all together..SIMCOM's onbaord GPS is pretty bad..paired a BN-220(Very small). GPS fix within 15 seconds , battery life has greatly improved.