Xinyuan-LilyGO/LilyGO-T-SIM7000G #4: T-SIM7000 evaluation and improvements

nikil511, 2020-02-05:
I have been playing with T-SIM7000 since December last year. We got some beta hardware, slightly different to the one on sale today.

I can confirm that everything works as they should, including NBIoT with PSM sleep (sub 1mA)
So overall very nice module, low power, connect to network even with battery at 2.8v.

Issues found so far:
1) was missing 100k resistor from BAT voltage divider (this could be because of beta hardware)

2) SOLAR input needs mask marks with +/- and its better to reverse the plug to be seed-studio polarity compatible, in my point of view

3) battery drain auto-boot seems ok but not perfect. It only pulls 35μΑ at 2.7V, when ESP32 and 3v3 is off, but when solar power is provided, and the battery has recovered, **sometimes** ESP32 boots in to "download mode" so it is working, but still needs a reset button :-(

lewisxhe, 2020-02-07:
The situation you said, we will test it, thank you for your feedback

nikil511, 2020-02-19:
@lewisxhe I performed more tests. The empty-battery-solar-charge boot problem happens only with one of the modules I have. The other one always boots fine. So maybe some component value is on the borderline.

lewisxhe, 2020-02-28:
I do n’t have T-SIM right now, I ca n’t do more tests. When T-SIM arrives, I will do more tests on power consumption. After the test is complete, I will summarize the summary of the next revision.

Mr-HaleYa, 2020-06-16:
1. was fixed
2. the markings were moved (the plug was not reversed 😞 but probably best bc my circuits would explode now)
3. this seems to be fixed with the new board (boot correctly when I charge from dead with solar)
4. same socket, unfortunately. they should have a more robust one 

If you feel your issue has been resolved, please close 🙂

lewisxhe, 2020-06-18:
The new board has been improved design, this issue will be closed

closed 2020-06-18 by lewisxhe

javipelopi, 2021-03-10:
Hi!

@nikil511 why do you say " Keep in mind that a solar bellow 1W will not charge the battery. (1W@5V = 80x100mm)." ?

Is it a limitation of cn3065 chip? Has it been solved in the final version?

Thanks in advance!