Xinyuan-LilyGO/LilyGO-T-SIM7000G #7: Network setup and TinyGSM AT command sketch

eiriksels, 2020-02-11:
I have not been able to set the network registration correctly to lock the board to NB-IoT mode. It seems to prefer 2G with the default sketches.

I want to set up: 
AT+COPS=1,2,”24202" (Norway teleoperator settings)

On previous boards(Sim7000 development boards) I have used the AT debug example within TinyGSM to set this up. I am not able to make this sketch work, even with adaptions. Any help on that would be appreciated.

I tried implementing the AT command directly to the TinyGSM Sim7000 library, but I receive only error back. Therefore I am now not sure how I should proceed to get my network setup correct.

eiriksels, 2020-02-11:
I was able to make the AT debug sketch function ok. I attach my sketch to here if anyone else needs it. It does not work perfect, but ok.

 I solved my connection issues by setting AT+CNMP to 38 and AT+CMNB to 2 (Forcing modem to LTE and NB-IoT network)

[AT debug.txt](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/4188267/AT.debug.txt)

Mr-HaleYa, 2020-05-09:
@eiriksels Where did you make the changes that forced it to use LTE?

eiriksels, 2020-05-13:
@Mr-HaleYa I used the sketch given in the AT debug.txt file attached and placed the AT commands as written there. After that it seems that the modem has been "remembering" the setting and have stayed at the NB-IoT net. http://www.microchip.ua/simcom/LTE/SIM7000/SIM7000%20Series_AT%20Command%20Manual_V1.04.pdf

You can also use the setNetworkMode(uint8_t mode) and  setPreferredMode(uint8_t mode) to do it directly in your sketch if you use TinyGSM library.

rakaandro, 2020-05-13:
I found a example for the NBIoT. It is using a modified TinyGSM Lib. I tested the 2G Mode, for the NBIoT Mode you do need a special SIM in Germany. They are no easy to get ;-(

[https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Platformio_Arduino_Thingsboard](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Platformio_Arduino_Thingsboard)

RaKa

Mr-HaleYa, 2020-05-13:
THANK YOU!!! I have been trying to get it to work for some time now. We are using a 7000G and need Cat-M1 connection and it would keep switching between that and 2g. Only problem with that is our 2g tower is getting removed at the end of the month lol (support isn't ending, it is literally getting bulldozed...)

Mr-HaleYa, 2020-06-06:
So your Sketch uses the startup of 
```cpp
  pinMode(PWR_PIN, OUTPUT);

  digitalWrite(PWR_PIN, HIGH);

  delay(10);

  digitalWrite(PWR_PIN, LOW);

  delay(1010); //Ton 1sec

  digitalWrite(PWR_PIN, HIGH);
``` 
but the example sketch uses
```cpp
    pinMode(PWR_PIN, OUTPUT);
    digitalWrite(PWR_PIN, HIGH);
    delay(300);
    digitalWrite(PWR_PIN, LOW);
``` 
why do you do it this way?
Is leaving the PWK_PIN HIGH going to cause problems or even work?

Mr-HaleYa, 2020-06-16:
@eiriksels I have solved my issue. If you have solved yours, please close

closed 2020-06-16 by eiriksels