Xinyuan-LilyGO/LilyGO-T-SIM7000G #10: SIM7000 nRESET overvoltage protection

ROSW6341, 2020-02-27:
The nRESET pin on the SIM7000 has a maximum input voltage of slightly over 2 volts.  The GPIO5 pin from the ESP32 can go as high as 3.3 volts.  Several options exist.  The first is to put a diode between the SIM7000 RESET pin and the rest of the circuit.  The second is to simply disconnect the pin completely from the circuit.  The SIM7000 hardware manual states "Note: This function is only used as an emergency reset. The RESET pin will be ineffectiveness in the power off mode.".  I've been testing several SIM7000 modules without ever needing to use the reset feature.  The third option is to leave the circuit as-is but state BOLDLY to configure the ESP32 GPIO5 as **OUT_OPEN_DRAIN**!
Personally I prefer option 2.  And it frees up a pin on the ESP32 which could possibly be used to switch power on & off to the GPS antenna circuit.  See my entry in the "Deep sleep power consumption" thread.

lewisxhe, 2020-02-28:
The SIM7000 reset pin will be removed. It will be used as a free GPIO in the next revision. I will test the antenna current. @ROSW6341

lewisxhe, 2020-02-28:
I do n’t have T-SIM right now, so I ca n’t do more tests, I ’m waiting for the board

Mr-HaleYa, 2020-06-16:
@ROSW6341 
A new version of the board has been released (20200415) Please have a look at the schematics

---
If you feel your issue has been resolved, please close  🙂

closed 2020-06-16 by ROSW6341