Xinyuan-LilyGO/LilyGO-T-SIM7000G #12: Incorrect VBAT reading on IO35 when on USB power

kapitan-u, 2020-03-03:
VBAT reading on IO35 is OK when running on battery power.
When USB power is connected for some reason it always reads 284 mV  (already x2)

T-18 board has no such problem.

lewisxhe, 2020-03-11:
In case of USB connection, the battery will be disconnected.

ro85ac, 2020-03-12:
> VBAT reading on IO35 is OK when running on battery power.
> When USB power is connected for some reason it always reads 284 mV (already x2)
> 
> T-18 board has no such problem.

@kapitan-u  can you paste the code of reading the battery power? Thanks

kapitan-u, 2020-03-12:
#include "driver/adc.h"
#include "esp_adc_cal.h"

#define DEFAULT_VREF 1100 // mV

uint32_t get_battery_voltage() {

	// ADC Configuration
	//
	adc1_config_width(ADC_WIDTH_BIT_10);
	adc1_config_channel_atten(ADC_BATTERY, ADC_ATTEN_DB_11);

	//Characterize ADC at particular atten
	//
	esp_adc_cal_characteristics_t *adc_chars = (esp_adc_cal_characteristics_t *) calloc(1, sizeof(esp_adc_cal_characteristics_t));
	esp_adc_cal_value_t val_type = esp_adc_cal_characterize( ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_10, DEFAULT_VREF, adc_chars);
    
	//Check type of calibration value used to characterize ADC
	//
	switch( val_type ) {

		case ESP_ADC_CAL_VAL_EFUSE_VREF:
			LOG(LL_INFO, ("ADC1 Callibration: eFuse Vref"));
			break;

		case ESP_ADC_CAL_VAL_EFUSE_TP:
			LOG(LL_INFO, ("ADC1 Callibration: Two Point"));
			break;
		
		default:
			LOG(LL_INFO, ("ADC1 Callibration: Default"));
			break;
	}

	uint16_t reading = adc1_get_raw( ADC_BATTERY );
	uint32_t voltage = esp_adc_cal_raw_to_voltage(reading, adc_chars) * 2; // *2 because of 100k/100k divider
	LOG(LL_INFO, ("Battery voltage: %u mV", voltage));
	
	return voltage;
}

ro85ac, 2020-03-13:
@kapitan-u I am new to ESP2 and unfortunately I cannot figure it out because ADC_WIDTH_BIT_10, ADC_WIDTH_BIT_10, ADC_ATTEN_DB_11 are not declared.

ro85ac, 2020-03-13:
@lewisxhe during the night, when the solar panel voltage is smaller than battery voltage, is there any  blocker to have only one-way (from panel to battery)?

lewisxhe, 2020-03-17:
Judging from the schematic, yes!

Mr-HaleYa, 2020-06-11:
@kapitan-u Close this if you have solved your problem

closed 2020-06-11 by kapitan-u