Xinyuan-LilyGO/LilyGO-T-SIM7000G #13: mqtt publish problem

crash007, 2020-03-06:
Hi,
can someone verify mqtt publish works with either DFRobot or TinyGSM?
Im unable to publish with both.

I can connect to host but when doing publish nothing happens. 
Im using GRPS mode and not NB.

from the server log:
1583473025: New connection from xx.xx.xx.xx on port 1883.
1583473025: New client connected from xx.xx.xx as T-Call-1 (p2, c1, k15, xxxx').
1583473025: No will message specified.
1583473025: Sending CONNACK to T-Call-1 (0, 0)
1583473047: Client T-Call-1 has exceeded timeout, disconnecting.

lewisxhe, 2020-03-11:
You can check out another warehouse with usage examples of TinyGSM and Cayenne.
https://github.com/Xinyuan-LilyGO/LilyGO-SIM7000-Cayenne

crash007, 2020-03-11:
Thanks I will check it out. So far I have tried with DFRobot ,TinyGSM and
Fona

crash007, 2020-03-18:
The problem was my mobile subscription. 
Can verify the mqtt examples work

closed 2020-03-18 by crash007