Xinyuan-LilyGO/LilyGO-T-SIM7000G #14: Firmware update -> sahara download not possible

kristo2301, 2020-04-13:
Hello, 

I have an issue to get HTTPs workign on module wit Firmware 03 so i would like to update FW to 04 but is not possible with QPTS -> getting sahara download error. -> following instrucition https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Firmware%20Updater%20Tool/How%20to%20Update%20SIM7000%20FW%20version%20with%20QPST.pdf

Is there any other way to update FW ?

Can you

Mr-HaleYa, 2020-06-16:
@kristo2301 a new version of the board 20200415 has been released and should have the up to date firmware otherwise no easy way to update it

--- 
if your issue has been solved please close 🙂 otherwise give more info

lewisxhe, 2020-06-18:
This requires a USB socket, soldered to the bottom of the board, corresponding to VCC D+ D- GND, I don’t know how you update it, do you update via USB or other methods?
The picture below is the connection method of the old version
[image]

kristo2301, 2020-06-18:
I tried USB methode but without success

Mr-HaleYa, 2020-06-18:
@kristo2301
 What exactly did you do? just saying you did it doesn't help much on our end of trying to troubleshoot.

kristo2301, 2020-06-18:
As in first post mentioned , acc to procedure and getting error

Mr-HaleYa, 2020-06-18:
@kristo2301 what version of the board do you have? 20191227 or 20200415

kristo2301, 2020-06-18:
20191227

Mr-HaleYa, 2020-06-24:
@lewisxhe 
can you tell us the procedure you go through when performing an update.

lewisxhe, 2020-06-24:
I will describe how to use USB update in my free time, but now I don’t have time

jrborbars, 2020-06-24:
I'm interested in this procedure too.

lewisxhe, 2020-07-06:
I found a video on the Internet to update the firmware. It should also apply to 7000G, but the firmware is different. Please see here https://www.waveshare.net/wiki/SIM7000-firmware-update-Video

Mr-HaleYa, 2020-07-13:
@kristo2301 @lewisxhe 
The answer has been shared and there has been no reply, this should be closed then reopened later if not working

Mr-HaleYa, 2020-07-26:
@lewisxhe can you like translate because I don't think anyone here reads Chinese? (i have no idea what language that even is)

eisaev, 2021-01-22:
I managed to restore the firmware of SIM7000G on the 20200415 board via USB pads after bricking the SIM7000 by running the command "AT+CFOTA=1". With the help of a pull-up resistor and a soldering iron, I succeeded.
But the only firmware I found is 1529B<b>01</b>SIM7000G (before that it was 1529B<b>04</b>SIM7000G). So I got a downgrade :(
Does anyone have more recent firmware? As far as I understand, the most recent one is 1529B<b>05</b>SIM7000G?

lewisxhe, 2021-03-01:
I have added the upgrade guide, please check here
[Upgrade guidelines](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/docs/How%20to%20update%20firmware.md)

lewisxhe, 2021-03-01:
@Mr-HaleYa @eisaev

closed 2021-03-01 by lewisxhe

rakaandro, 2021-03-07:
Hello,
the download link to firmware 1529B05SIM7000G seems to be dead. Can you fix it, thanks.

lewisxhe:
>>I have added the upgrade guide, please check here
>>Upgrade guidelines

jrborbars, 2021-03-26:
The Linux instructions (on PDFs contained in the .rar file) only contain chinese words. Can you (or other) provide in plain English? Thank you.

barbiani, 2023-03-25:
> This requires a USB socket, soldered to the bottom of the board, corresponding to VCC D+ D- GND, I don’t know how you update it, do you update via USB or other methods? The picture below is the connection method of the old version [image]

I have done it, but no USB devices are recognized. The modem is working.

Is there anything else needed to be done to put in boot mode?