Xinyuan-LilyGO/LilyGO-T-SIM7000G #15: Send mail over smtp and battery deep discharge protection

vladco78, 2020-05-01:
Hi everybody,
I can not send email over smtp, failure 65. Witch the same code on SIM808 is all ok. Has anyone tried this yet?

Is it correct that the module did no battery deep discharge protection?  What is the simplest solution?

Mr-HaleYa, 2020-06-16:
1. Can you post the code you are trying to use?

2. the old version of the board (you can see the number written on it 20191227) did not have over-discharge protection but the new version 20200415 does have that as well as overcharge protection

vladco78, 2020-06-17:
I use these commands, it works fine on a SIM808 module!

AT+SAPBR=2,1
+SAPBR: 1,1,"XX.XX.XX.XX"
OK

AT+EMAILCID=1
OK

AT+EMAILTO=60
OK

AT+SMTPSRV="smtp.1und1.de",25
OK

AT+SMTPAUTH=1,"login","pw"
OK

AT+SMTPFROM="mail",""
OK

AT+SMTPRCPT=0,0,"mail recipient",""
OK

AT+SMTPSUB="subject"
OK

AT+SMTPBODY=258

DOWNLOAD
DATA************************
OK

AT+SMTPSEND
OK

+SMTPSEND: 65

lewisxhe, 2020-06-18:
I haven't tested the mail function, I will test it.

lewisxhe, 2020-06-18:
After checking SIM7000G does not support mail function

Mr-HaleYa, 2020-06-18:
@vladco78 
since this can not be fixed and all we can do is recommend a different board OR a different protocol this can now be closed

vladco78, 2020-06-18:
OK, good to know! Thank you for your support!

Mr-HaleYa, 2020-06-18:
@vladco78 
You have to push the close button 🙂

closed 2020-06-18 by vladco78