Xinyuan-LilyGO/LilyGO-T-SIM7000G #16: Use Cat-M1?

Mr-HaleYa, 2020-05-09:
So it appears that all the examples use GPRS?
We couldn't find any examples using 4G, LTE, or Cat-M1.
From what I have read and been told GPRS is 2g (or sometimes called 2.5G). Our State is ending all 2G support very soon and 3G will follow quickly after. We have been trying to find a way to use Cat-M1 with the TTGO sim7000g but could not find any documents explaining how. The SIM7000G documentation shows that it supports the Cat-M1 protocol.

Is a special sim card required to use Cat-M1? How do you know if you are connected over Cat-M1 and not 2G? Is this even supported by any available libraries?

Any help with this would be greatly appreciated. 

We posted this and an issue because we are having issues with getting it to work and found no examples using it.

lewisxhe, 2020-05-10:
You can try to use this library . https://github.com/botletics/SIM7000-LTE-Shield

Mr-HaleYa, 2020-05-10:
Are you saying that you have never tested Cat-M1 support on the device that you created?

lewisxhe, 2020-05-11:
You can watch here  ,
https://simcom.ee/documents/SIM7000E/SIM7000%20Series_AT%20Command%20Manual_V1.03.pdf 

 These are the supported modes
[image]

Mr-HaleYa, 2020-05-11:
Yes, I have a copy of the same pdf printed and was reading over the commands. I just want the modem to use LTE and *not* 2G. Is this possible with the examples you've given? What mode is it set to by default and where is that set?

Also, do you know what is 5.2.19 AT+CMNB Preferred selection between CAT-M and NB-IoT?

lewisxhe, 2020-05-11:
If you use TinyGSM, you can use modem.getNetworkModes () to get the current network, use modem.setNetworkMode (38) to use LET,
Use modem.setPreferredMode (1), use CAT-M

lewisxhe, 2020-05-11:
See AT + CMNB for specific meaning

lewisxhe, 2020-05-11:
Oh, I was wrong. Use AT + CNMP? To check the currently selected network

Mr-HaleYa, 2020-05-11:
OK thank you so much. I will test this as soon as my boards arrive (slow shipping...) and I will let you know if this works.

Mr-HaleYa, 2020-06-07:
Ok so I got the ttgo sim7000G (finally) and i am running it with most of my old code jut with the new startup sequence. My question is. How do I know it is actually using CAT-M1??? 
```cpp
  modem.setPreferredMode(38);// 2 Automatic / 13 GSM only / 38 LTE only / 51 GSM and LTE only
  modem.setPreferredMode(1); // 1 CAT-M / 2 NB-Iot / 3 CAT-M and NB-IoT
```
I have these set before `!modem.waitForNetwork` as is in the example. It connects to something and works because I can send data to our server. How do I know its catm? Is it guaranteed to be since i set those parameters?

lewisxhe, 2020-06-18:
You can check before use to see if the return is what you need, such as 
```
AT+CNMP?
```
 Get current network

Mr-HaleYa, 2020-06-24:
ok, thank you for all your help. I will reopen if I am unable to get it working

Mr-HaleYa, 2020-07-15:
Ok I have been doing a lot of testing and I have a Hologram SIM that supports all network types. I have the code set that i listed before
``` cpp
  modem.setPreferredMode(38);// 2 Automatic / 13 GSM only / 38 LTE only / 51 GSM and LTE only
  modem.setPreferredMode(1); // 1 CAT-M / 2 NB-Iot / 3 CAT-M and NB-IoT
```
after these run, I check with AT+CNMP? (I get 38) and AT+CMNB? (I get 1)

The problem is it never connects... I did have it connect once but I went to the hologram dashboard and it said it was connected over 2G. I don't know what I'm doing wrong or what to even do...

lewisxhe, 2020-07-16:
This question, I will test in free time

Mr-HaleYa, 2020-07-16:
@lewisxhe 
I can ONLY get it to connect with 2g no matter what AT commands I do or what libraries I use... it's like this thing is unusable. 

I can't find a single guide for the sim7000g that connects to the network and works... I really need help with this. I have been trying for months now and I still am stuck with 2g. I have a Hologram sim and Twilio sim just to see if changing companies would fix it but no. Please do help

lewisxhe, 2020-07-16:
Okay, I will test it tomorrow, please wait

lewisxhe, 2020-07-16:
Sorry, I just found out that in my country, there is no CAT-M1 base station deployed. I cannot test it. I will ask SIMCom if I want to test

Mr-HaleYa, 2020-07-16:
can you test NB-IOT?

lewisxhe, 2020-07-16:
Of course, I can test it, I remember what I tested, and I will reply to you after testing again

Mr-HaleYa, 2020-07-16:
if you can test it and show me how and I can get it to work I will buy 2 more then next month 10 more

lewisxhe, 2020-07-16:
I just tested it, NB mode is normal,Please see the screenshot below
[image]
[image]

lewisxhe, 2020-07-16:
Do you use a dedicated NB-IOT SIM card? If you use a normal SIM card, you will not be connected to the NB network.

Mr-HaleYa, 2020-07-16:
I use a sim from hologram.io 

> Our single SIM card works in every available country and with all of our partner carriers. Connect across every technology, including 2G, 3G, 4G LTE, CAT-M, and NB-IoT.
You don't need special SIM cards to access the latest network technologies. As Hologram's network expands and improves, your SIM automatically upgrades to support new services, carriers, and technologies.

Mr-HaleYa, 2020-07-16:
what ide is that??? can you post the full code so I can copy lol. I do the same commands and wait for OK but when it use to connect it was 2G. Now it won't even connect to a network

lewisxhe, 2020-07-16:
The code is to use the sample code, I use vscode, platformio plugin

Mr-HaleYa, 2020-07-16:
I know I can access the NB network it says on the package of sims 2G, 3G, 4G LTE, CAT-M, and NB-IoT

Mr-HaleYa, 2020-07-16:
you can test the `AT+CPSI?` command or any command with this 

```cpp
    int i = 10;
    while (i) {
      SerialAT.println("AT+CPSI?");
      delay(100);
      if (SerialAT.available()) {
        String r = SerialAT.readString();
        Serial.println(r);
        if ( r.indexOf("OK") >= 0 ) break;;
      }
    delay(900);
    i--;
    }
```

Mr-HaleYa, 2020-07-16:
So I left it trying to connect with the settings at 38 LTE and 3 Cat or NB for **2 hours** and got this as an output

``` cpp
[663] Wait...
[3663] Initializing modem...
[14859] ### Unhandled: f`f
[14859] ### TinyGSM Version: 0.10.5
[14859] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000
[15174] ### Modem: SIMCOM SIM7000G
[15192] Modem Name: SIMCOM SIM7000G
[15198] Modem Info: SIM7000G R1529
[16205] ### Unhandled: OK           <-- sets to 38
[17706] ### Unhandled: OK           <-- sets to 3
[18206] Waiting for network...
[88294] Waiting for network...
```
the same message repeating so skip a bunch of time to here
``` cpp
[8148568] Waiting for network...
[8197278] Network connected           <--- IT SAYS CONNECTED??
[8197278] Connecting to hologram
[8269696] Waiting for network...
[8340446] Waiting for network...
[8341447] ### Unhandled: +CME ERROR: no network service    <- What is this?
[8410495] Waiting for network...
[8480583] Waiting for network...
[8531361] Network connected         <--- IT SAYS CONNECTED??
[8531361] Connecting to hologram
[8607114] Waiting for network...
[8677890] Waiting for network...
[8678891] ### Unhandled: +CME ERROR: no network service    <- What is this?
[8747938] Waiting for network...
[8818026] Waiting for network...
```
whole bunch more time, skip to here
``` cpp
[11271141] Waiting for network...
[11341229] Waiting for network...
[11368176] Network connected         <--- IT SAYS CONNECTED??
[11368176] Connecting to hologram
[11441814] Waiting for network...
[11512564] Waiting for network...
```
 and it was still doing this when I stopped it
``` cpp
[14736658] Waiting for network...
[14806747] Waiting for network...
[14876835] Waiting for network...
```

Mr-HaleYa, 2020-07-17:
WHAT THE HECK????

I literally only changed the **_ANTENNA_** and now

``` cpp
[3663] Initializing modem...
[15505] ### Unhandled: @
[15505] ### TinyGSM Version: 0.10.5
[15505] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000
[15955] ### Modem: SIMCOM SIM7000G
[16118] Modem Name: SIMCOM SIM7000G
[16157] Modem Info: SIM7000G R1529
[16746] ### Daylight savings time state updated.
[16747] ### Network time and time zone updated.
[18721] ### Unhandled: OK
[19221] Waiting for network...
[19301] Network connected
[19301] Connecting to hologram
[23204] ### Daylight savings time state updated.
[23205] ### Network time and time zone updated.
[23375] GPRS status: connected
[23422] CCID: 89445023032060*****
[23462] IMEI: 8699510311*****
[23522] Operator: AT&T Hologram
[23568] Local IP: 10.117.120.155
[23606] Signal quality: 17

+CPSI: LTE CAT-M1,Online,310-410,0x9709,111159055,393,EUTRAN-BAND12,5110,3,3,-16,-113,-83,10

OK

[27753] GPRS disconnected
```

**LTE CAT-M1,Online**

___

Every time I use the included full band LTE antenna it does not ever work except 2G, Now I'm using a LORA 868-915MHz 2dBi Gain, and it's connecting to CAT-M1!!!

**PLEASE START SELLING WITH ONE OF THESE!!!** 😭😭😭 
I know they make a LOT of these in China so it should be included with (Or replace) the full band antenna. I think it should have both so you can try either one.
[image]

___
My code... just took TinyGSM example and slimmed it down to just GPRS
``` cpp
#define TINY_GSM_MODEM_SIM7000

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

/*
   Tests enabled
*/
#define TINY_GSM_TEST_GPRS true
#define TINY_GSM_POWERDOWN true

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[]  = "hologram";
const char gprsUser[] = "";
const char gprsPass[] = "";

#include <TinyGsmClient.h>
#include <Ticker.h>

TinyGsm modem(SerialAT);

#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  30        /* Time ESP32 will go to sleep (in seconds) */

#define UART_BAUD   9600
#define PIN_DTR     25
#define PIN_TX      27
#define PIN_RX      26
#define PWR_PIN     4
#define LED_PIN     12

void setup()
{
  // Set console baud rate
  SerialMon.begin(9600);
  delay(10);

  // Set LED OFF
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(300);
  digitalWrite(PWR_PIN, LOW);

  DBG("Wait...");

  delay(3000);

  SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  DBG("Initializing modem...");
  if (!modem.restart()) {
    DBG("Failed to restart modem, delaying 10s and retrying");
    // return;
  }

  String name = modem.getModemName();
  DBG("Modem Name:", name);

  String modemInfo = modem.getModemInfo();
  DBG("Modem Info:", modemInfo);

#if TINY_GSM_TEST_GPRS
  // Unlock your SIM card with a PIN if needed
  if ( GSM_PIN && modem.getSimStatus() != 3 ) {
    modem.simUnlock(GSM_PIN);
  }
#endif

  /*
    2 Automatic
    13 GSM only
    38 LTE only
    51 GSM and LTE only
  * * * */
  String res;
  do {
    res = modem.setNetworkMode(38);
    delay(500);
  } while (res != "OK");

  /*
    1 CAT-M
    2 NB-Iot
    3 CAT-M and NB-IoT
  * * */
  do {
    res = modem.setPreferredMode(1);
    delay(500);
  } while (res != "OK");

}

void loop()
{

  DBG("Waiting for network...");
  if (!modem.waitForNetwork()) {
    delay(10000);
    return;
  }

  if (modem.isNetworkConnected()) {
    DBG("Network connected");
  }

#if TINY_GSM_TEST_GPRS
  DBG("Connecting to", apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    delay(10000);
    return;
  }

  bool res = modem.isGprsConnected();
  DBG("GPRS status:", res ? "connected" : "not connected");

  String ccid = modem.getSimCCID();
  DBG("CCID:", ccid);

  String imei = modem.getIMEI();
  DBG("IMEI:", imei);

  String cop = modem.getOperator();
  DBG("Operator:", cop);

  IPAddress local = modem.localIP();
  DBG("Local IP:", local);

  int csq = modem.getSignalQuality();
  DBG("Signal quality:", csq);

  delay(2000);
  int i = 10;
  while (i) {
    SerialAT.println("AT+CPSI?");
    delay(500);
    if (SerialAT.available()) {
      String r = SerialAT.readString();
      Serial.println(r);
      if ( r.indexOf("OK") >= 0 ) break;;
    }
    delay(500);
    i--;
  }
#endif

#if TINY_GSM_TEST_GPRS
  modem.gprsDisconnect();
  if (!modem.isGprsConnected()) {
    DBG("GPRS disconnected");
  } else {
    DBG("GPRS disconnect: Failed.");
  }
#endif

#if TINY_GSM_POWERDOWN
  // Try to power-off (modem may decide to restart automatically)
  // To turn off modem completely, please use Reset/Enable pins
  modem.poweroff();
  DBG("Poweroff.");
#endif

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  delay(200);
  esp_deep_sleep_start();

  // Do nothing forevermore
  while (true) {
    modem.maintain();
  }
}
```

lewisxhe, 2020-07-17:
Hi, you mean that you can use CAT-M after replacing the antenna?

This is the feedback I got from using AT+CPSI? to prove that I belong to the NB network

[image]

Mr-HaleYa, 2020-07-17:
Yes after changing to the new antenna it works. I still can't get NB-IoT to work tho... It just says Waiting for network... But i don't need NB I need CAT-M1.

Anyways I have got it to work with CAT-M1 finally!!! The included Antenna does not work at all for me... it only connects to 2G sometimes, most of the time it can't connect to anything but the LORA antenna that I got with an ESP32 LORA works perfectly. you really should start selling with one included

lewisxhe, 2020-07-17:
I can use NB with the attached antenna, but I cannot test CAT-M1, I will report the problem to LilyGO

Mr-HaleYa, 2020-07-17:
@lewisxhe
 Thank you so much for all the help ❤️

lewisxhe, 2020-07-17:
@Mr-HaleYa Can you tell me the frequency band you are using? I am asking about the antenna provider.

lewisxhe, 2020-07-17:
I saw that you use AT++CPSI?
Frequency Band = 5110?

Mr-HaleYa, 2020-07-17:
for AT+CPSI? the AT doc says
`+CPSI: <System Mode>,<Operation Mode>,<MCC>-<MNC>,<TAC>,<SCellID>,<PCellID>,<Frequency Band>,<earfcn>,<dlbw><ulbw>,<RSRQ>,<RSRP>,<RSSI>,<RSSNR>
`
and i get 
`+CPSI: LTE CAT-M1 , Online , 310-410 , 0x9709 , 111159055 , 393 , EUTRAN-BAND12 , 5110 , 3 , 3 , -16 , -113 , -83 , 10`

So it looks like `EUTRAN-BAND12`

Mr-HaleYa, 2020-07-21:
now it's not working again and since that day it has never connected again. It will not connect to ANYTHING now...

I described the issue here
https://github.com/vshymanskyy/TinyGSM/issues/421

Mr-HaleYa, 2020-07-29:
I have fixed this, will reopen if new issues arise

closed 2020-07-29 by Mr-HaleYa