Xinyuan-LilyGO/LilyGO-T-SIM7000G #17: Specification incomplete - solar voltage input range?

Elmi77, 2020-05-15:
As is can't find this information anywhere: what is the allowed input voltage range at the JST solar input connector?

Thanks!

nikil511, 2020-05-19:
v1 that is available at the time of writing this uses TP4056
ABSOLUTE MAXIMUM RATINGS ·Input Supply Voltage(VCC)：-0.3V～8V 
But for solar to charge the batteries you need >4.5v 
long story short, use a **5.5v or 6v panel**

Elmi77, 2020-05-25:
Cool, thanks!

ROSW6341, 2020-05-28:
Foe protection the TP4056 has a 6.8v zener diode across the input from the solar panel.  Most nominal sized panels will have a noticeable voltage drop when charging the battery.  The voltage from my 4"x6" 6 volt panel drops to below 4 volts in all but very bright sunshine.  I decide to use a panel designed to charge a 6v SLA battery.  It has an unloaded output of over 9v.  Instead of imposing a linear regulator, I chose a step down switching regulator with an output voltage of 5v.  The excess voltage is converted of available current by the regulator.  I found complete modules on Amazon at 5 for $8.

ROSW6341, 2020-05-28:
Correction, the modules I purchased were $9 for 5 and I have not tested them in the field.  There are other modules on Amazon at lower prices.  One that shows up frequently is labeled Mini360.  In perusing information about it, I came across a review where the author did extensive testing and found they were not very efficient. <https://goughlui.com/2018/07/04/tested-mini-360-mp2307-based-3a-buck-converter-module/>.

Mr-HaleYa, 2020-06-11:
@Elmi77 Close this if you have solved your problem

Mr-HaleYa, 2020-06-17:
@Elmi77 
The new board (20200415) uses a CN3065 chip to charge. Looking at the schematics there is no protection IC for it so the charging range is now 4.4v to 6v

I have submitted a PR that adds this info to the readme so others will know

---

If You have solved your issue please close 🙂

Mr-HaleYa, 2020-06-24:
@Elmi77 @lewisxhe 
The question has been answered and added to the readme
this can now be closed

closed 2020-06-24 by lewisxhe