Xinyuan-LilyGO/LilyGO-T-SIM7000G #18: Sleep whole board?

Elmi77, 2020-05-25:
I try to send my board to deep sleep (with wakeup by timer and external pin) using following code:

```
   // GPS
   modem.disableGPS();
   // Set SIM7000G GPIO4 HIGH ,Close GPS power
   // CMD:AT+SGPIO=0,4,1,0
   modem.sendAT("+SGPIO=0,4,1,0"); 

   //GSM
   modem.poweroff();

   esp_sleep_enable_ext0_wakeup(GPIO_NUM_33,1);
   esp_sleep_enable_timer_wakeup(100000*uS_TO_S_FACTOR); // sleep for 1.15 days
   esp_deep_sleep_start();
```

But after that a green LED near the SIM-socket is still lit and a red one right beside it still flashes. What is missing here, what do I need to turn off? The "all functions" example shows only one LED at GPIO 12.

Thanks :-)

lewisxhe, 2020-05-26:
This is a serial port indicator. When using USB power supply, it will not be controlled and will not light up when using the battery.

Elmi77, 2020-05-26:
OK, I found there is an other heartbeat-LED right beside the GPS antenna connector - it still flashes also when I call modem.sendAT("+SGPIO=0,4,1,0"); to close GPS power. Shouldn't this be off too?

lewisxhe, 2020-05-27:
GPS power control does not have this function in the current version. Next to GPS is the status indicator, which will be turned off after calling modem.poweroff ();

closed 2020-05-28 by Elmi77