Xinyuan-LilyGO/LilyGO-T-SIM7000G #19: Q10 Failure

kapitan-u, 2020-05-25:
One of my LilyGO-T-SIM7000G has failed. It used to work from battery power and charging battery from USB. I never user solar charging port. Now it do not work from battery anymore while it is still operational while powered from USB.  Also there is no VBAT voltage on pin 1 expansion header.
VCC_BAT is OK, so I'm suspecting Q10 failure. Please investigate design, it would be a drag if such failure will be occurring while board is in mass production.  Thank you!

lewisxhe, 2020-05-26:
I have feedback and I am waiting for the result!

Mr-HaleYa, 2020-06-12:
@kapitan-u what version of the board did you have this happen on?

kapitan-u, 2020-06-12:
V1.0

Mr-HaleYa, 2020-06-17:
@kapitan-u 
so if you were using the 20191227 then its kind of too late since they are no longer in production and a new version of the board has been made hopeful correcting this. The most that could be done is adding a warning on the examples page explaining not to do as you did.

___

If your issue is resolved, please close 🙂

kapitan-u, 2020-06-17:
Please elaborate on: "hopeful correcting this".
And: "explaining not to do as you did".
In my opinion I did nothing wrong.
Thank you!

Mr-HaleYa, 2020-06-17:
Hopefully, it's fixed in the new board revision, I have not had it happen and I've shared my batteries with USB.

Well if charging with USB broke it we can put a warning saying don't charge the batteries with USB on the old board. I'm not saying you where wrong I'm just saying it obviously broke something so maybe they shouldn't do that if they don't want it to possibly happen to them.

lewisxhe, 2020-06-18:
The new board has improved solar design, this issue will be closed

closed 2020-06-18 by lewisxhe