Xinyuan-LilyGO/LilyGO-T-SIM7000G #20: (Re)Setting power of SIM-module has no effect

Elmi77, 2020-06-01:
Hi,

in example code here from GitHub for initialisation following sequence can be found:

```
#define PWR_PIN     4

    pinMode(PWR_PIN, OUTPUT);
    digitalWrite(PWR_PIN, HIGH);
    delay(300);
    digitalWrite(PWR_PIN, LOW);
    delay(3000);
```

Next the GSM modem is initialised and used. From what can be seen in code, one could assume digitalWrite(PWR_PIN, HIGH) turns off the power for the modem and digitalWrite(PWR_PIN,LOW) turns it back on. But it seems this has absolutely no effect: when first setting it to LOW and then to HIGH (which should be power off), no difference can be noticed, the modem initialises in the same way as before.

So what is wrong here? Is it possible switching the power does not work at all? Or is the wrong pin defined for power?

lewisxhe, 2020-06-05:
This is the PWR_KEY Pin of SIM7000, you can also control it to get the effect of power on and off, please refer to the manual

Elmi77, 2020-06-05:
OK but the point is: it does not work! No matter what value the GPIO is set to, the modem always works in the same way, so it seems it does not switch any power!?

lewisxhe, 2020-06-06:
If you do not want to use the command to shut down, you can pull down PWR_PIN and keep it low for 1.2 seconds, then pull it high to force shutdown

like this 
digitalWrite(PWR_PIN,LOW);
delay(1300);
digitalWrite(PWR_PIN,HIGH);
[image]

Mr-HaleYa, 2020-06-11:
@Elmi77 close this if you have solved your problem

closed 2020-06-14 by Elmi77