Xinyuan-LilyGO/LilyGO-T-SIM7000G #21: AllFunctions example not working, getting brownout triggers

ticketos-pl, 2020-06-03:
Just got my t-sim7000g board, wanted to give it a simple test, but I'm failing on running the AllFunctions example from this repo.

Strange thing - I got modem info returning data only once.
```
Console output:
[15528] ### Unhandled: f`f
[15528] ### TinyGSM Version: 0.10.5
[15528] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000
[15843] ### Modem: SIMCOM SIM7000G
[15870] Modem Name: SIMCOM SIM7000G
[15876] Modem Info: SIM7000G R1529
[16882] ### Unhandled: OK
[18383] ### Unhandled: OK
[18883] Waiting for network...
[29307] ### Unhandled: f`f
[48161] ### Unhandled: f`f
[67538] ### Unhandled: f`f
[89055] Waiting for network...
[101103] ### Unhandled: f`f
[117336] ### Unhandled: f`f
[135141] ### Unhandled: f`f
[160480] Waiting for network...

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
SDCard MOUNT FAIL
[2044] Wait...
[5044] Initializing modem...
[15045] ### Unhandled: f`f
[15045] Failed to restart modem, delaying 10s and retrying
[16045] Modem Name: SIMCom SIM7000
[17045] Modem Info: 
[24045] ### Unhandled: f`f
[30047] Waiting for network...
[39112] ### Unhandled: f`f
[55084] ### Unhandled: f`f
[76820] ### Unhandled: f`f
[102160] Waiting for network...

Brownout detector was triggered
```

.... and so on

I have the same situation when using http client example from tinyGsm repo:
https://github.com/vshymanskyy/TinyGSM/blob/master/examples/HttpClient/HttpClient.ino

Console output:
```
Initializing modem...
[17445] ### Unhandled: f`f
Modem Info: 

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Wait...
Initializing modem...
```

Any ideas? I also have t-sim800l board, and I didn't have such issues.

slavino, 2020-06-04:
Just to make sure as you are referring to SIM800 - are you using NB-IoT SIM card or common LTE/4G?

ticketos-pl, 2020-06-04:
Sorry it turned out to be a low quality USB-C adapter I was using with my mac. That's why I had brownout triggered. 

Changed to a different adapter and now the example code is running without any issues :)

Mr-HaleYa, 2020-06-11:
@ticketos-pl close the issue if you fixed the problem

lewisxhe, 2020-06-18:
If there are still problems, please reopen the issue

closed 2020-06-18 by lewisxhe