Xinyuan-LilyGO/LilyGO-T-SIM7000G #23: What are the different types????

Mr-HaleYa, 2020-06-12:
Ok so now I have 3 different boards that have shown up at my house... I have a 20191211 a 20191227 and a 20-4-15. **WHATS THE DIFFERENCES????** and why is there no documentation showing different types and new/removed futures???

Seriously post some documentation explaining these! or even explain it to me and ill make a PR idc but people need to know this stuff. also, schematics and dimensions are different so that needs to be updated as well.

nikil511, 2020-06-12:
> me and ill make a PR idc but people need to know this stuff. also, (issue
> @lewisxhe <https://github.com/lewisxhe> @nikil511
> <https://github.com/nikil511> @ROSW6341 <https://github.com/ROSW6341> You
> three seem like the best to answer this question...

:-) in addition I have some beta that never made it to production. So yes,
its a mess.
I have some old schematics but I cant guarantee they match the PCBs.

I guess @lewisxhe <https://github.com/lewisxhe> can help eh?
(else I will start digging my files)

lewisxhe, 2020-06-12:
Sorry, the documentation hasn't been updated yet, these boards will be explained and updated on Monday.  As for the difference, as far as I know, there should be only two versions. The second version is to solve the problem of excessive current consumption of the GPS in the first version, and there are some small changes. I need to ask them because there are too many boards, I don’t remember  live.

lewisxhe, 2020-06-15:
The schematic and related descriptions have been updated. I can't do anything about version management. This is not something I can control. Although I have feedback many times, it is still so confusing. I used the date of the board as the version number. @Mr-HaleYa

Elmi77, 2020-06-15:
> I can't do anything about version management. This is not something I can control.

To make it easier: how can one recognise the latest version of the board?

Mr-HaleYa, 2020-06-15:
@Elmi77 the numbers writhen on the board in center right. The new one starts with 2020

closed 2020-06-15 by Mr-HaleYa