Xinyuan-LilyGO/LilyGO-T-SIM7000G #25: No GSM network functionality - waitForNetwork() never succeeds

Elmi77, 2020-06-15:
I'm initialising all the GSM-stuff based on the code out of the all-functions-example. Unfortunately this does not work, waitForNetwork() never returns with true. When calling modem.getSignalQuality() right before waiting for the network, it shows me a signal quality of 99. What I already have tried:

- the board is battery-backed, so there should be enough power
- the SIM card is verified to work (done with an Android device)
- I tried a second antenna as replacement for the original one
- I added a call modem.sendAT("+CFUN=1") right after modem.restart() to ensure radio is powered

All this did not help. So: what could be the reason for the troubles? What could I try in order to get it working?

rakaandro, 2020-06-15:
I made a echo terminal for the TTGO-SIM7000G, you can send AT commands over USB-Serial to the modem.

check if PIN is okay
check available networks and so on ..... The is a .pdf out there with all the commands.

```
/***
           Echo terminal for TTGO T-SIM7000G
           open USB-COM Port with a terminal programm and send AT commands
           directly to Sim7000G (over the ESP32)

           by RaKa 06/2020          
 ***/

#include <HardwareSerial.h>

// modem connections
#define MODEM_TX 26
#define MODEM_RX 27
#define MODEM_RST 5

void setup()
{
  Serial.println("Modem hardware reset first ....");
  pinMode(MODEM_RST, OUTPUT);
  digitalWrite(MODEM_RST, LOW);
  delay(260); //Treset 252ms
  digitalWrite(MODEM_RST, HIGH);
  delay(4000); //Modem takes longer to get ready and reply after this kind of reset vs power on
  Serial.println("Modem hardware reset ready!");

  //start serial connection
  Serial.begin(115200);

  //Serial1 at pin 12 and 13
  Serial1.begin(9600, SERIAL_8N1, MODEM_TX, MODEM_RX);
  Serial.println("ready. Enter AT commands:");
}

void loop()
{
  int data;
  //wait for terminal commands
  while (Serial.available())
  {
    // send data to Serial1 (SIM7000G)
    data = Serial.read();
    Serial1.write(data);
  }

  //wait for data from Serial1 (SIM7000G)
  while (Serial1.available())
  {
    //send received data to terminal
    data = Serial1.read();
    Serial.write(data);
  }
}

```

lewisxhe, 2020-06-16:
What frequency band SIM card are you using?
The default example uses setNetworkMode, 51 as the preferred parameter. Does your country still support 2G? I tried sometimes it is difficult to connect using 51, and it takes a long time to connect, if your country still supports 2G, you can change 51 to 13 to try!

Mr-HaleYa, 2020-06-16:
America has killed 2G (T-Moble is the last with it but they are quickly phasing it out). T-moble is also shutting down 3G so you may be having problems based on carrier

Elmi77, 2020-06-16:
> I tried sometimes it is difficult to connect using 51, and it takes a long time to connect, if your country still supports 2G, you can change 51 to 13 to try!

13 seems to work. Amazingly the description in example code says this is GSM!?
I hadit set to 2 (automatic) but it seems the automatism isn't that automatic :-D

Mr-HaleYa, 2020-06-17:
@Elmi77 
Ya, the boards can be finicky sometimes and that is mostly attributed to how the tinyGSM library functions. 

---

If you have solved your problem, please close 🙂

closed 2020-06-17 by Elmi77