Xinyuan-LilyGO/LilyGO-T-SIM7000G #26: General question about solar

Mr-HaleYa, 2020-06-15:
First question is what is the operating voltages of the solar panel? I have 5 volt panels, 6 volt panels and 12 volt panels. What is the recommended voltage and wattage?

Elmi77, 2020-06-15:
According to https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/17 input voltage should be less than 8V, but I'm not sure if something has changed with the new 2020 hardware revision.

Elmi77, 2020-06-15:
...and please keep in mind the maximum voltage of a solar panel is always bigger than the nominal voltage!

Mr-HaleYa, 2020-06-16:
Its 5.5v to 6v If anyone else is looking for this

closed 2020-06-16 by Mr-HaleYa

ro85ac, 2020-07-22:
@Mr-HaleYa I am using the device with a 6W 6V  solar power.  Data every 15 seconds to the cloud by wifi. As you can see in the attached picture, I cannot understand why the battery is not charging when there is voltage in the solar panel. 

Do you have any ideea?

[image]

Mr-HaleYa, 2020-07-23:
I believe it has mppt power tracking which basically only charges when the sun is at an optimal angle using math to calculate the best voltage because I'm like a battery or solar panel can only Supply a certain amount of current so you have to charge when it's at its highest voltage and current or else you draw too much power and the voltage will drop below what it can actually be charged with. Something like that...

I use an external solar charger because my panels are 12-volt and don't work with the 6-volt input and I get the exact same thing because that's how the solar chargers work it is very bad for your batteries to sit at the max capacity for an extended period Of they will literally explode.

As long as it is charging to the 4.2 max voltage at least once a day or whenever you need it then it's it should not be sitting at 4.2. It should hit 4.2 and then tomorrow charge back to 4.2

[image]

ro85ac, 2020-07-23:
> I believe it has mppt power tracking which basically only charges when the sun is at an optimal angle using math to calculate the best voltage because I'm like a battery or solar panel can only [...]

The solar panel stays during day at 4.2V and drops to 0V during night, but the battery keeps discharging. I am using the cayenne project. Am I using a wrong solar panel or the board is not working properly ( I have an oldder version --- the one from January )?

Mr-HaleYa, 2020-07-23:
@ro85ac 
Are you saying that it never fully recharges and instead is slowly dropping over time?

ro85ac, 2020-07-23:
@Mr-HaleYa 
Yes, as you can see on the first graphic, the battery never recharges. It keeps dropping.

Mr-HaleYa, 2020-07-23:
can you please share a picture of the solar panel with something easy to compare sizes such as a US quarter  or a soda can

ro85ac, 2020-07-23:
@Mr-HaleYa 
This is the link where I bought the solar panel. It's the model 6V 1000mA size (170mm x 230mm)
https://www.aliexpress.com/item/32877917510.html

Mr-HaleYa, 2020-07-23:
hmmm I have some similar panels that are 6v and they charged mine in a couple of hours and I had 4 18650s in parallel so the capacity was much higher. How many panels do you have?

ro85ac, 2020-07-23:
@Mr-HaleYa I have one battery and one solar panel. [image]

Mr-HaleYa, 2020-07-23:
please continue this on #47