Xinyuan-LilyGO/LilyGO-T-SIM7000G #28: Use external battery...

Mr-HaleYa, 2020-06-16:
Ok, so I desoldered the battery-pack that is on the board and was trying to use one that has 3 x 18650 in parallel just for more capacity.
How do you connect this correctly? do I need to solder my backs terminals to the + - tabs on the back of the PCB or can I use the Vbat pin and gnd?

I tried using the Vbat pin connected to my batteries + and gnd connected to the - and it would flash the blue LED then turn off (I think it was off but my script shuts it off as soon as it boots so idk) none of the other listed turned on ( I only kept the battery in for ~3 sec to hopefully not kill it if I was doing it wrong) I toggled the power switch and pressed the PWR button and nothing happened...

Sadly, I tried to flip the battery to see if that was the problem and I got Magic Blue Smoke from the TCS9208 so that is dead now... I have 5 more boards but I don't want to do that again so I am asking here

Mr-HaleYa, 2020-06-16:
oh my gosh, I'm stupid... I had it... I took a look at the schematics and the green and red led don't tun on when using the battery and the Vbat and the + on the bat back are the same pin............ 😭😭😭

Mr-HaleYa, 2020-06-16:
Is there any way for you board makers to add backward battery protection and led PLEASE it would have saved me a board... I guess I can still use the USB-c connection...

Mr-HaleYa, 2020-06-16:
I've replaced my board with a new one and Is there 5v power on the VIN pin when using Battery because none of my attached sensors are working and I can't get a read with a multimeter.

Mr-HaleYa, 2020-06-16:
So There is no 5v output if you are using a battery??? how am I supposed to use anything? what are these even made for if they can't even power a sensor to collect data? This has to be a mistake right? (3.3v isn't enough to power for what I'm doing I need 5v)

@nikil511
you seem to have experience with these boards, what is your opinion? shouldn't it have a 5v output?

lewisxhe, 2020-06-17:
VIN is a USB input. If battery power is used, there will be no voltage output from VIN. It is only input and cannot be output

Mr-HaleYa, 2020-06-17:
@lewisxhe 
Thank you. After reviewing the schematics I came to the same conclusion. I was able to change out my sensors for 3.3v tolerant ones.

closed 2020-06-17 by Mr-HaleYa