Xinyuan-LilyGO/LilyGO-T-SIM7000G #29: Further source of information?

grichfitz, 2020-06-16:
Hya,
I’m about to upgrade a project from the T-Call to this T-SIM. 
There is a lot of information available on the web for the T-Call, not so much for the T-SIM, because it so new. 
Has anyone else found a website with more info than that is on here??
TIA

Mr-HaleYa, 2020-06-16:
@grichfitz lol nope, this is it 😄 I can try to help you if you would like. What questions do you have?

grichfitz, 2020-06-16:
Hello Mr HaleYa,
Thanks for your reply. I wasn’t expecting a questions and answer, but you help will be most helpful. 

1. Can I power the board via 5v pin? And will this charge the battery? Will the board reset if power supply removed and it switches to the battery?
2. My project uses ESP_SLEEP_WAKEUP_EXT1, on pins, 12, 13, 14, 15, 25. This seems to be the max on the T-Call which gives me feedback to which pin woke it up. What pins can I use on the T-SIM? I see some of these are taken up by the SIM7000 chip. 

Any help with this will be greatly appreciated. 

TÍA grichfitz

Mr-HaleYa, 2020-06-16:
@grichfitz 
Yes, you can power with 5v on the VIN pin (bottom right) looking at the schematics it will charge the battery. It should transfer seamlessly but I have never tried.

I use pins 18, 19, 23, 35 (25 is used by DTR but can be used if you don't require DTR which is used to wake the modem if you put it to sleep or shut it down) I'm sure more pins can be used but I have not tested others

grichfitz, 2020-06-16:
Mr HaleYa,
Your a gentleman with information, thanks.
I´ll be testing this at the weekend when the board arrives.
Thanks

Mr-HaleYa, 2020-06-16:
just letting you know... DO NOT PUT THE BATTERY IN BACKWARDS!!!!! I killed mine... #28 

___
If your issue has been resolved please close 🙂

grichfitz, 2020-06-17:
What version board do you have? Version 20200415 is meant to have protection against this.

Mr-HaleYa, 2020-06-17:
@grichfitz 
ha no it doesn't... Maybe if use the battery socket, but I was using an external battery pack (3x 18650 in parallel for more capacity) connected to VBAT and GND and it was all cool and dandy until I put the battery in backward then poof magic blue smoke and battery IC no longer works lol

It does have reverse solar panel connection if you connect your solar panel backward if that's what you were thinking

___

any devs reading, I think you should redo the vbat pin so it's connected in the same fashion that they positive terminal of the battery socket is connected as in you can use the on-off switch and it uses all the same circuit as the integrated battery socket. also, throw in some backward battery protection if you can.

Mr-HaleYa, 2020-06-24:
@grichfitz 
refine your question, ask more questions, or can you close this since it has now gone a week with no activity?

closed 2020-06-24 by grichfitz