Xinyuan-LilyGO/LilyGO-T-SIM7000G #32: Status LED / Status Signal

kapitan-u, 2020-06-20:
It will be very useful to have SIM7000 status signal connected to ESP32 gpio and to remove status LED.
Booth measures will improve battery life.

Mr-HaleYa, 2020-06-24:
How can you connect it to the esp32 and remove it? Wouldn't that just remove it altogether?  🙃 

In order to make it function as I believe you are saying. You want the esp32 to decide whether or not to allow the signal light to turn on or off, is not an easy task because you can either have the esp32 read the status of the Sims 7000g which I do not know of any library that can do that easily quickly or efficiently and then toggle the light or have the esp32 block the Flow by activating a transistor or deactivating it so that the same 7000g cannot toggle the light. 

If you are that concerned about battery life take a soldering iron and pop the LED off lol

I did find it strange that they added to LEDs for Sim 7000g status in the new version. I guess the left one tells you whether it has power and the right one tells you it's connectivity status to GSM. I guess if you have GSM disabled but you want to know if the modem has power then they both come into use. For example, if you were to shut down the GSM connection when you're not using it it would still have power but the GSM indicator LED would not be blinking. I actually wish the old version was like the new version because I would shut down my GSM connection and that was the only LED on the board so then I didn't know if the board had power or not. it was installed outside on a solar panel with batteries and so I would have to sit there for 10 minutes and wait for it to reconnect or I would have to go inside and get my probe to check voltages.

lewisxhe, 2020-06-24:
This opinion is used in T-PCE. For T-SIM, the next improvement will be made after the board is sold out
The network status light can be turned off, as described here
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800L/issues/72#issue-639741563

Mr-HaleYa, 2020-06-24:
@lewisxhe 
How many are left and what are you wanting to change in the next version? I want to use these as a mainboard in around 100+ devices so I'm worried if big changes are coming...

lewisxhe, 2020-06-24:
If there are no major problems with customer feedback, we will not make major changes. If you have any requirements, you can ask them.

Mr-HaleYa, 2020-06-24:
I would really really like an improved solar charger because even with the solar charger that is in it right now I'm using a independent solar charger because it is very hard to find large solar panels that stay within the charging range. I'm using 10-watt 12 volt panels. I tried a 6 volt 5 watt panel but each panel was plus or minus 5% of that voltage and some of them were frying the boards charging circuit because they were spiking up to 7.4 volts and if the battery was already charged that voltage couldn't go anywhere and would fry the battery charger. I understand improving this would add cost but it is very necessary cost because the solar charger is not very useful in large applications it's only useful in very small garden panels for testing in your window as if it's not very robust

lewisxhe, 2020-06-24:
So, your question now is that the solar voltage input range on the board is too low?

Mr-HaleYa, 2020-06-24:
I will open an issue in the morning regarding a full explanation of what I mean and my requirements if our team is going to buy  over a hundred of these we are honestly hoping to get around 1000 within the year

Mr-HaleYa, 2020-06-24:
@kapitan-u  @lewisxhe 
I think this can be closed as the question was answered and alternatives where given

closed 2020-06-28 by lewisxhe