Xinyuan-LilyGO/LilyGO-T-SIM7000G #33: Instructions to setup the Arduino IDE?

Humancell, 2020-06-24:
Where can I find the instructions for setting up the Arduino IDE to program this device?

Humancell, 2020-06-24:
Here is an update:

Following the instructions here:  https://sites.google.com/site/jmaathuis/arduino/lilygo-ttgo-t-display-esp32

I did add the ESP32 Boards file from here: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

Now I do see a long list of ESP32 compatible boards ... but not this one!

Which one do I select?  Do I follow the instructions in the link above and select "ESP32 Dev Module" and the rest of this parameters?

Mr-HaleYa, 2020-06-24:
select esp32 dev board at the top... then select the port it's connected to and that's all, nothing fancy. 

[image]

Mr-HaleYa, 2020-06-28:
@Humancell  @lewisxhe 
This can be closed as it's not an issue but rather a lack of understanding how to use Arduino IDE. Also the solution was given

closed 2020-07-01 by Humancell