Xinyuan-LilyGO/LilyGO-T-SIM7000G #34: Questions and maybe some requests

Mr-HaleYa, 2020-06-24:
As I said in a previous issue post, My team and I are looking to buy 100+ of these boards.

I would really really like an improved solar charger because even with the solar charger that is in it right now I'm using an independent solar charger because it is very hard to find large solar panels that stay within the charging range. I'm using 10-watt 12-volt panels. I tried a 6-volt 5-watt panel but each panel was plus or minus 5% of that voltage and some of them were frying the boards charging circuit because they were spiking up to 7.4 volts and if the battery was already charged that voltage couldn't go anywhere and would fry the battery charger. I understand improving this would add cost but it is a very necessary cost because the solar charger is not very useful in large applications it's only useful in very small garden panels for testing in your window as if it's not very robust.

I see that there is not much room left on the PCB without either switching components to the bottom or increasing the size of the PCB.  It looks like some resistors could be moved to the left side of the left and right of the esp32 and the sim7000g Gould be moved down ~3mm. That would make some room for additional voltage control hardware that would greatly improve usability. 

We use an mppt that has a 28v range with a 3.7v output to charge our 18650s 
https://www.aliexpress.com/item/32862470804.html?spm=a2g0s.13010208.99999999.321.189c3c003qqGg6
[image]

lewisxhe, 2020-06-26:
I'm not an in-depth user of this board, I can only return your feedback.

Mr-HaleYa, 2020-06-28:
that would be great if you could advise them to allow a higher range of charging input via solar.

Mr-HaleYa, 2020-07-01:
If you have anything else to add, please comment, otherwise I hope you tell the engineers. I will close this since it is technically, not an issue.

closed 2020-07-01 by Mr-HaleYa

fgnievinski, 2023-06-06:
@Mr-HaleYa @xyzzy42: based on your experiences, where would be the best point in this Lilygo board to connect an external solar charger -- the Vin pin or the battery connectors? 

We're using the [Adafruit solar charger](https://www.adafruit.com/product/4755) whose "Load output is regulated to never be over 4.4V so it's safe for 3.3V regulators or 5V boost converters". 

We're considering the following two scenarios:
1. with a a step up in Vin:
[image]
2. without a step up, in the battery connectors on the back of the board ([not the Vbat/GND pins](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/40#issuecomment-660769873)):
[image]

the first scenario works but it seems wasteful considering the series of [down regulators](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/237#issuecomment-1576129143) in the Lilygo board.

Mr-HaleYa, 2023-06-06:
@fgnievinski You should not comment on _old **CLOSED**_ issues... Nobody likes [Nercoposting](https://www.urbandictionary.com/define.php?term=necropost)...

___

The first scenario is a waste of a regulator.
The second is close but you would be putting 4.4v into where a 3.7v battery should be and that's bad...
Instead, connect the Lipo pad and GND on the adafruit board to where the battery would be on the back of the TTGO board.
That way the TTGO board thinks there's just a battery there but the battery is also attached to the solar charger.

I love adafruit, I worked with them and their products are excellent. 👌
I use [these ](https://www.aliexpress.us/item/3256803364381619.html) solar chargers in my production boards. They work amazingly and are very affordable

xyzzy42, 2023-06-06:
@Mr-HaleYa, do you mean here?
[image]

I wouldn't do that.  The load output from the adafruit charger board would then be connected to the battery protection circuit, which limits voltage to 4.2 V.

I'd connect it to the VBAT pin.  And always have the battery switch on the lilygo board set to the OFF position.

The power switch has the 18650 battery, the battery protection circuit, and the battery charger on one side, and the VBAT pin and everything else on the other.  So by going to the vbat pin and keeping the switch off, it avoids the battery protection and charger circuit, which aren't desirable with the load output from the adafruit charger board.

The 4.4V limit on the charger board load output is higher than the boost regulator voltage of 4.2V, but it should handle it fine and the 3.3V linear regulators shouldn't have a problem with the higher voltage either.

fgnievinski, 2023-06-06:
I think @Mr-HaleYa's recommendation is NOT to use the Adafruit charger's load output at all -- maybe something like this?

[image]

the recommendation above seems to agree with Adafruit's tech support, which made no big deal about skipping the load sharing feature (however, that was for an earlier version of the charger, whose output was NOT regulated):

https://forums.adafruit.com/viewtopic.php?p=337687#p337687

I'll also check @xyzzy42's suggestion, of connecting the Adafruit charger's load output to Vbat/GND pins on the TTGO; but, if I'm not mistaken, we have burned a board that way in the past (although I don't remember the details, maybe a USB connector was being used in conjunction).

thank you both for your input (and sorry for any disturbance)!

Mr-HaleYa, 2023-06-06:
@xyzzy42 not the charger load output... the lipo itself. if you're going to do it that way cut the switch off in the OFF position because if it's ever bumped ON it would probably kill the board, at least the charge circuit. 

> NOT to use the Adafruit charger's load output at all

@fgnievinski Yes, like that. This is the way I do it on all boards, works great. Doing it this way you can use USB at the same time. just make sure the power switch is OFF.

xyzzy42, 2023-06-07:
Ok, connecting the battery directly would not have the same issue with the lilygo battery protection circuit trying to "protect" the regulated solar panel output from the adafruit charger.

The adafruit charger is a much better charger than the one on the lilygo board.  If you connect the battery directly you will lose some features.  The adafruit charger will not cycle the battery if it is fully charged and the panel can supply the needed load on its own.  This is better for battery life.  Drawing a load from the battery directly will also mess up the charge termination current.  Charging stops when the current drops below a certain level.  If the load is coming directly from the battery, there is no way for the charger to tell if charge current is going to the battery or to the load.

The reasoning for not connecting the load output of the charger to the powered device appears to have been solely because that version of the board could drive the load output to 6V+.  This would indeed be a big problem.  But the new version of the board is limited to 4.4 V and that is not much more than the 4.2 V battery limit and should be within the maximum ratings of the boost regulator and the linear regulators after it.

xyzzy42, 2023-06-07:
> Doing it this way you can use USB at the same time. just make sure the power switch is OFF.

How do you figure that?  The schematics place the charger and the battery, `VCC_BAT`, on the same side of the switch, pin 2.  So connecting USB will charge the battery with the switch in either position.

fgnievinski, 2023-06-07:
thank you @xyzzy42, I follow your reasoning. ideally we'd take advantage of load sharing, so that solar is prioritized over battery when available, to avoid cycling the battery. here's a diagram illustrating my interpretation of your proposed setup:
[image]
now the battery is kind of far from the load, so maybe it'd be good to use an extra capacitor, to smooth out the ~2A spikes?