Xinyuan-LilyGO/LilyGO-T-SIM7000G #35: sim7000g unable to connect when sun is in shadow

crash007, 2020-06-26:
When connected to solar panel (5v, 100ma) and the sun is in shadow sim7000g is unable to connect to network.
If i turn solar panel upside down(solar vcc is close to 0) or when strong sunlight it will connect and work. 
With solar panel disconnected it also works well
This is old model Version 20191227.
I have tried with 470 capacitor GND VBAT with no success.

lewisxhe, 2020-06-26:
The new version has solved this problem.  As far as the version is concerned, this problem cannot be solved very well.

crash007, 2020-06-26:
Okey thank you. I did a new order of 4 pcs June 03 will I receive the new version?

lewisxhe, 2020-06-26:
The old version has been eliminated, and the new version is now on sale.  So you will receive the new version, in order to make sure, you better ask customer service.

Mr-HaleYa, 2020-06-28:
@crash007 @lewisxhe 
this can be closed as the issue has been resolved

closed 2020-06-28 by lewisxhe