Xinyuan-LilyGO/LilyGO-T-SIM7000G #36: Green and red leds are ON at same time, what's wrong?

procopio, 2020-06-27:
Hi,
I have 5 boards testing an kind of GPS Tracker.
One of those board are not working well, AT commands are slow to answer and green and red led are ON.
Please can you tell me what could be wrong with the board and if I can do something to repair it?
Thanks,
Joao

[image]

Mr-HaleYa, 2020-06-28:
@procopio 
 just confirming: Red light does not flash at all? it is just solid?
Did you connect a battery in wrong? I know this can fry the charge circuit which is what the Green and Red LEDs are connected to. It only takes an instant to fry the components then they will misbehave like this. you may also just have a defective board as the first version was not the best...

My best guess is that you accidentally fried the TP4056 either by attaching the solar panel backwards (as the first person did not have protection from this) or connecting the battery backwards or possibly just a static shock when you went to picked it up which you may not have even felt because it doesn't take much of a shock to kill sensitive Electronics. I would just use the other boards that you have and toss this one out

procopio, 2020-07-01:
I tried to change the TP4056 and nothing changed.
I will not lose my time with that damaged board. 
I will keep testing the other 4.
Closing the ticket.
Tks

closed 2020-07-01 by procopio