Xinyuan-LilyGO/LilyGO-T-SIM7000G #37: Does not recover after battery dies

Mr-HaleYa, 2020-07-05:
I am using the latest version of the board and have an external battery pack of 18650s that are wired in series attached to the vbat pin and ground. I am using in mppt solar charger to charge the battery pack that is attached because the onboard solar charger has too tight of voltage tolerance. 

I have been using it without any problems for months, during the day the charger charges the batteries and at night the batteries keep  everything going.

Anyways, I wanted to do a test to see what would happen if the solar panel got covered in dirt or if it rained for a week. I disattached the solar panel and let the batteries drain until the board shut off. When I measure the battery's voltage they were at 1.7 volts which I expected they would be lower than the 2.5 volt cut off because I have an external sensor attached that can draw voltage all the way down to 1.7 volts... I then reattached the solar panel and left it charging in the sun. After about six hours I was getting suspicious since I had not seen any new data up here on my graphs. I went outside and check the voltage of the batteries and they were at 3.7 volts but the SIM 7000 status LEDs were off.

The blue LED was on and solid. my sketch shuts the blue LED off after boot to save some power. No other LEDs were on except the blue one. I press the power button and nothing happened I was only able to get it to recover by pressing the reboot button.

I **need** these boards to be able to recover after batteries died and are recharged. Is this a known issue? could it be caused because I am using an external battery pack connected to the vbat pin and gnd instead of the positive / negative terminal of the battery connector that is on the back of the board? Does the board have any kind of circuitry that is supposed to have it recover such incidents? Or can it be added?

lewisxhe, 2020-07-06:
I think it should enter the download mode unexpectedly. I suspect that the download circuit affects it. If you can, please remove them and try again. The previous version also reported this problem.

Mr-HaleYa, 2020-07-06:
Do what?

Mr-HaleYa, 2020-07-06:
I _need_ to be able to do updates on these, They will be installed in remote locations and I will be doing GSM OTA updates, since that is software (I assume) modifying the hardware won't affect that?

lewisxhe, 2020-07-08:
Hi , @Mr-HaleYa 
I personally think it may accidentally enter the download mode, so you must manually reset it. If you can, I ask you to remove the download circuit. Of course, this is just my guess.

Mr-HaleYa, 2020-07-08:
I put a dead battery in a different board with the exact same configuration and this time when it started charging with solar it recovered... I have some batteries on a test station so that they drain till they're dead so I can do a couple more of these test runs but it appears that it's a 50-50 chance now.

Also can you take a picture of the board and circle what I'm supposed to remove or cut or modify because I don't really want to start playing doctor with it when I don't know what I'm messing with

lewisxhe, 2020-07-08:
Although I can’t be 100% sure, I will give you the original part of the screenshot download part tomorrow, if you can, you can remove it and try again, but it cannot be automatically downloaded after removal.

lewisxhe, 2020-07-09:
Hi @Mr-HaleYa 
If you can, please remove the components marked in the figure below, I think it has a lot to do with it. Note that it cannot be automatically downloaded after removal. You can short IO0, then reset, and then click download.

[image]

Mr-HaleYa, 2020-07-09:
Can this be fix in the future? I really like the idea of adding a switch to "allow" programming or not so that this issue never happens if the switch is off (disconnects this circuit)

lewisxhe, 2020-07-09:
It is temporarily unclear that it is caused by it, this only happens in some cases, you can try it first, and then feedback to us

Mr-HaleYa, 2020-07-11:
I have tried with another dead battery and it recovered fine... I don't want to destroy one of my boards for no reason lol. Can the engineers test it since they should be looking to fix such problems?

I think you might be able to connect to a power supply and slowly turn up the voltage to mimic solar charging.

Mr-HaleYa, 2020-07-15:
What appears to happen is that the board is trying to power bellow a capacity (not voltage but capacity) that can not be sustained... the solar charger puts out the required voltage amount to charge the batteries, and the board sees this voltage and tries to turn on, then idk what happens but I assume the mppt cant supply the right amount (it's rated to charge output at 3 amps so should be plenty...) or it shuts off or something because the board appears to shut off then repeats... it looks to be strobing extremely quickly, but that is only with a dead battery... after the battery charges this should stop, but it doesn't. the led that I have programmed to turn off after successful boot (onboard led blue) never shuts off and the board never recovers until RST pressed

nissimzur, 2020-07-17:
Slow voltage power up may not generate a reset. Many MCU has this problem. 
I offer to stay in light sleep, no GSM, test voltage and send esp_restart(); when voltage is high to normal boot the GSM chip.
Another option to use a MOSFET that will block the power to PCB below 3.7V. So, PCB will get powers only when the battery if full to work. GSM pulls a lot of power in powerup.

Mr-HaleYa, 2020-07-17:
So have it boot and check the voltage and run a low power function that uses nothing then once the voltage is high have it reboot and run normally?

Mr-HaleYa, 2020-07-18:
> Another option to use a MOSFET that will block the power to PCB below 3.7V. So, PCB will get powers only when the battery if full to work. GSM pulls a lot of power in powerup.

@nissimzur 
What kind of circuit would this look like?

Mr-HaleYa, 2020-07-18:
@lewisxhe 
shouldn't this not be happening because there is a 2.4v voltage cutoff for battery discharge protection. when I plug in a 3.2-volt battery it doesn't turn on but if start charging it starts trying to turn on... shouldn't the voltage cutoff stop this until a voltage is met?

Could it be spikes of voltage from the solar charger making it think it's high enough so it tries to boot? Would adding a large smoothing cap help? 

would adding a MOSFET even help? would it be "tricked" by charging spikes? 

when I connect battery to charger and meter to battery I don't see any spikes but they would be happening really fast because the blue light flickers until ~2.6v the goes solid (board turning on and locking)

Mr-HaleYa, 2020-07-19:
@lewisxhe 
oh my gosh, I think I found the solution!!! https://4donline.ihs.com/images/VipMasterIC/IC/FAIR/FAIRS25678/FAIRS25678-1.pdf?hkey=EC6BD57738AE6E33B588C5F9AD3CEFA7

as I was saying before I would do Pin1 to Batt, pin2 to GND, and Pin3 to RST so that when the voltage is below the threshold on Pin1 then Pin3 will be held LOW making the board be held in RST so it will not boot then once the voltage is above the threshold Pin3 will be held HIGH allowing the board to boot normally!!!

I think the KA75250 or KA75270 would be PERFECT since they are 2.5v or 2.7v thresholds. 

They make these as tiny ICs (SOT-89) so this could easily be added to the board in the next revision so that they never have this horrible problem again! have one between the VBAT, GND, and RST so it stays off till VBAT is above the threshold value you choose. The IC comes on at 0.8V so it will be the first thing on, before the MCU.

Please let me know your thoughts!

Mr-HaleYa, 2020-07-19:
Those components are hard for me to find in the USA ao maybe this will work better. It is smaller and lower power with a threshold of 2.7v. NCP301HSN27T1G https://www.onsemi.com/pub/Collateral/NCP300-D.PDF

Mr-HaleYa, 2020-07-20:
The only thing that I see that could be a problem is that the SIM7000G still activates when the RST is LOW and the Arduino is off so I don't know if it can get stuck in a locked state like the MCU.

lewisxhe, 2020-07-29:
When RST is at low level, it can only make ESP32 unfair, if the voltage can make SIM7000G start, then SIM7000G will still work normally, if the battery is not enough, then SIM7000G will automatically shut down

Mr-HaleYa, 2020-07-29:
@lewisxhe  did you show the engineer what I found? I think adding that part would eliminate the MCU startup error.

Mr-HaleYa, 2020-08-15:
If you use the battery pad on the back it will not run into this problem, but if you use the Vbat pin to power the board you can have the MCU brick. The Vbat pin needs to be rebuilt to be the same as the pad on the back of the PCB, otherwise its literally useless because it has no protection features and can brick the MCU on low power boot.

Closing this since it will require a new revision of the board to correctly fix

closed 2020-08-15 by Mr-HaleYa