Xinyuan-LilyGO/LilyGO-T-SIM7000G #38: Modem of module not responding on AT commands

cuplesoft, 2020-07-06:
Hello,
I bought this module for tests and at the beginning everything was fine : network connection, gps data etc.

After two days of tests I lost connection with modem. I can't get any response on AT commands.
I may by possible that firmware was broken or SIM7000G died..?
Can I upload default firmware for this module and bring my module back to life?

Mr-HaleYa, 2020-07-06:
@cuplesoft 
You probably just messed up the 7000g. this can happen if you are messing with AT commands.

this is on the TinyGSM page

> Sometimes (especially if you played with AT commands), your module configuration may become invalid. This may result in problems such as:
> 
> Can't connect to the GPRS network
> Can't connect to the server
> Sent/received data contains invalid bytes
> etc.
> To return module to Factory Defaults, use this sketch: File -> Examples -> TinyGSM -> tools -> [FactoryReset](https://github.com/vshymanskyy/TinyGSM/blob/master/tools/FactoryReset/FactoryReset.ino)

___ 

You need to modify it to look like this 

```
// Select your modem:
#define TINY_GSM_MODEM_SIM800

#define MODEM_RST     5
#define MODEM_PWKEY   4
#define MODEM_DTR     25
#define MODEM_TX      26
#define MODEM_RX      27

#include <TinyGsmClient.h>

// Set serial for debug console (to the Serial Monitor, speed 9600)
#define SerialMon Serial

// Set serial for AT commands (to the module)
#define SerialAT Serial1

#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);

void setup() {
  // Set console baud rate
  SerialMon.begin(9600);
  delay(10);

  // Set GSM module baud rate
  pinMode(MODEM_PWKEY, OUTPUT);
  digitalWrite(MODEM_PWKEY, HIGH);
  delay(10);
  digitalWrite(MODEM_PWKEY, LOW);
  delay(1010); //Ton 1sec
  digitalWrite(MODEM_PWKEY, HIGH);
  delay(4510); //Ton uart 4.5sec but seems to need ~7sec after hard (button) reset, On soft-reset serial replies immediately.

  // Set and start the GSM module at baud rate
  SerialAT.begin(9600, SERIAL_8N1, MODEM_TX, MODEM_RX); //reversing them
  delay(6000);

  bool ret = modem.factoryDefault();

  SerialMon.println(F("***********************************************************"));
  SerialMon.print  (F(" Return settings to Factory Defaults: "));
  SerialMon.println((ret) ? "OK" : "FAIL");
  SerialMon.println(F("***********************************************************"));
}

void loop() {

}
```

I have it set to sim800 because the commands are __very__ similar (and they dot have sim7000 set up yet so it won't work 🙃 ) 

your output in serial should be 

```
AT+IPR=0
AT+IFC=0,0
AT+ICF=3,3
AT+CSCLK=0
AT&W
AT&W

OK
***********************************************************
 Return settings to Factory Defaults: OK
***********************************************************

```

I have ran this sketch on the 20200415 version of the board and it completes successfully. If this does not fix your problem let me know, also tell me the board version (20191227 or 20200415)

best of luck 
~Hale

cuplesoft, 2020-07-06:
Thanks for nice solution but it doesn't work on my device :(

Return settings to Factory Defaults: FAIL

It's look like modem starting normally : STATUS led is on, NETLIGHT led starts blinking once per second and several seconds after blinks once per 3 seconds. This is correct behavior?

But output from modem doesn't exists - requests only:

[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
SDCard MOUNT FAIL
[2068] Wait...
[5069] Initializing modem...
AT+CFUN=0
怘⸮f`⸮f⸮⸮[15070] ### Unhandled: f`f
[15070] Failed to restart modem, delaying 10s and retrying
AT+GMM
[16070] Modem Name: SIMCom SIM7000
ATI
[17070] Modem Info: 
AT+CPIN?
AT+CPIN?
AT+CPIN?
AT+CPIN?
AT+CPIN?
AT+CNMP=51
AT+CMNB=2
[30072] Waiting for network...
AT+CEREG?
AT+CGREG?
AT+CEREG?
...
AT+CGREG?
AT+CEREG?
AT+CGREG?
[100850] Waiting for network...
AT+CEREG?
AT+CGREG?
AT+CEREG?
AT+CGREG?
AT+CEREG?
...

Mr-HaleYa, 2020-07-07:
I ran the script that I shared on 5 of my boards and it finished with `Return settings to Factory Defaults: OK` on all of them. If you can not get it to run then nothing will...

Mr-HaleYa, 2020-07-07:
have you used battery power or just USB? if you put the battery in backward it can cause "undesired behavior" and also will kill the charging circut...

cuplesoft, 2020-07-08:
I used many configurations - tests ;)
Generally I connected my module to my Laptop. Of course it many times with battery NCR18650B or with battery only.
Charging circuit you said... Hmm, this may be possible because at the center of module green led is on, BUT near of green is red led and blinking very fast (5-10 blinks per second) This is not too good i think..

lewisxhe, 2020-07-08:
Have you tested whether the AT has responded using serial port loopback?  Also, on the board, there seems to be two LED lights near SIM7000. Is it normally on?

Mr-HaleYa, 2020-07-08:
that is normal for the green light to be solid in the red light to be blinking extremely quickly. that means it is working correctly with USB. When connected using the battery pack on the back those lights function as charge indicators for solar panels. But when using USB they don't have actual functionality they just blink as energy pulses through the charge circuit

lewisxhe, 2020-07-09:
The green light is the full indicator
The red light is the charging indicator,
When it is not connected to the battery, the status is undefined.

You can try to connect the battery and observe whether the indicator light is normal.

The indicator lights I mentioned are two red lights next to the SIM7000G. One silk screen is NETLIGHT, STATUS. Please check if STATUS is on.

Mr-HaleYa, 2020-07-09:
@cuplesoft 

These LEDs at the bottom
[image]

cuplesoft, 2020-07-09:
Finally! My device is working now :-)

I did it by running this code:

[https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/ResetModem](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/ResetModem)

In my case BOUND must be 9600 and after upload I need to reset module manually.
Not every time this procedure works because sometimes result was "Reset settings to Factory Default: FAIL". 
I reset manually by button again and now works!

Thanks for help for everyone.

lewisxhe, 2020-07-09:
It may be that you set the baud rate, if it is not the default, then it is changed, and then you send any command will not return.

closed 2020-07-09 by lewisxhe

philo-create, 2020-07-20:
Hello.i have  the same  problem  with  my  board.

> Hello,
> I bought this module for tests and at the beginning everything was fine : network connection, gps data etc.
> 
> After two days of tests I lost connection with modem. I can't get any response on AT commands.
> It may be possible that firmware was broken or SIM7000G died..?
> Can I upload default firmware for this device and bring my module back to life?

I have  used the  recommended  code to reset the modem and got the  following  output but my  modem still doesn't work,what else do you recommend  i should do
`***********************************************************
 Return settings to Factory Defaults: OK
***********************************************************`

philo-create, 2020-07-20:
Hello.i have the same problem with my board.

> Hello,
> I bought this module for tests and at the beginning everything was fine : network connection, gps data etc.
>
> After two days of tests I lost connection with modem. I can't get any response on [...]

I have used the recommended code to reset the modem and got the following output below but my modem still doesn't work,what else do you recommend i should do?

`***********************************************************
 Return settings to Factory Defaults: OK
***********************************************************`

Mr-HaleYa, 2020-07-20:
what did you do, to begin with???

philo-create, 2020-07-20:
the board has been working fine. I have  been using  it to read  humidity and  temperature  data for a month now  but  then it suddenly stopped connecting  to the network,just woke-up one-day and  the  board wasn't  working  properly anymore.

Mr-HaleYa, 2020-07-20:
what connection typer were you using(2-3-4G catm nb) and what sim card? and where are you located

philo-create, 2020-07-21:
The board had been working for a month  over the 4g connection before it suddenly stopped working ,i have another similar board that i got the same time that is still working,  therefore the problem has nothing to do with the sim card  (I tested the sim card to make-sure) or location (South Africa).

Mr-HaleYa, 2020-07-21:
when you run the reset example do you have ATdebug enabled. if so can you share the serial output

philo-create, 2020-07-21:
So just run  the  reset  example and i am  getting  a  different  output,before  my  output was  successful but  today its a  fail

[image]

philo-create, 2020-07-21:
forgot to  mention  that i am  using  the 20191227 Version ,i know  that this  board  doesn't have  the   reverse battery protection and  that  can  affect the modem,so  i made  sure  that the  problem wasn't t battery related before  i posted my  problem  here

Mr-HaleYa, 2020-07-21:
sometimes it fails sometimes it passes its dependent on the modems mood... press the rest on the board until you get a pass and send me a pic of that if you can

pixelfriese, 2021-07-31:
Hi. I have the same problem with my board too.

When I run the reset program, I get the following output. The baud rate is set correctly. I can't send AT commands to the board from the console. Only when I insert the AT command in the program code will they run. My guess is that I won't get any feedback on the if (SerialAT.available) query if the modem is already running.:

```
14:25:32.941 -> 
14:25:32.941 -> Starting Up Modem...
14:25:32.941 -> 
14:25:32.975 -> Testing Modem Response...
14:25:32.975 -> 
14:25:32.975 -> ****
14:25:34.437 -> u⸮
14:25:36.443 -> AT

14:25:36.477 -> OK
14:25:36.477 -> 
14:25:36.477 -> ****
14:25:36.477 -> 
14:25:36.477 -> AT&FZE0&W
14:25:36.545 -> AT&FZE0&W

14:25:36.613 -> Autobaud support: (9600,19200,38400,57600,115200).
14:25:36.613 -> 
14:25:36.613 -> OK
14:25:36.613 -> AT+IPR=0
14:25:37.567 -> AT+IFC=0,0
14:25:38.587 -> AT+ICF=3,3
14:25:39.573 -> AT+CSCLK=0
14:25:40.593 -> AT&W
14:25:41.751 -> 
14:25:41.751 -> ***********************************************************
14:25:41.751 ->  Reset settings to Factory Default: FAIL
14:25:41.751 -> ***********************************************************

```

But if I press the reset button on the board again and the modem is already active, I get the following output:

```
14:32:37.269 -> Starting Up Modem...
14:32:47.517 -> 
14:32:47.555 -> Testing Modem Response...
14:32:47.555 -> 
14:32:47.555 -> ****
14:32:57.745 -> ****
14:32:57.745 -> 
14:32:57.745 -> ***********************************************************
14:32:57.745 ->  Failed to connect to the modem! Check the baud and try again.
14:32:57.745 -> ***********************************************************
14:32:57.745 -> 
```

cuplesoft, 2021-11-29:
I'm here again...
Why blue led light is always on when usb is connected?

nkkn87, 2023-04-20:
Hi, I understand that this post was long time ago. But now I am using SIM7000E and use to send data to server with http get. SIM could stop working (power light is on but newtwork status light totally off), So I cannot run continuously. When this happens, I cannot send any commands, even AT command. I have to power cycle the SIM module and it starts working again. 
Anybody encountered this and any idea how to fix this?

ceren16, 2023-08-09:
> sometimes it fails sometimes it passes its dependent on the modems mood... press the rest on the board until you get a pass and send me a pic of that if you can

Hello my board is not still working , how can I solve it?