Xinyuan-LilyGO/LilyGO-T-SIM7000G #40: Questions about board (20200415)

Mr-HaleYa, 2020-07-10:
So I hooked a battery to the vbat pin and gnd backwards and this component exploded....

What is it? Why did it? isn't there supposed to be backwards battery protection or is that only if you use the socket on the back?

[image]

lewisxhe, 2020-07-10:
Sorry, I will report the situation to the hardware engineer, and I will ask them what to do.

slavino, 2020-07-10:
> Sorry, I will report the situation to the hardware engineer, and I will ask them what to do.

maybe bigger "BAT +" and "BAT -" signs on the board in next HW version :D

Mr-HaleYa, 2020-07-10:
It was an external battery pack, but yes they could put bigger symbols on the back

lewisxhe, 2020-07-11:
Did you use the rear 18650 battery holder to access the battery before? Is it normal ?

Mr-HaleYa, 2020-07-11:
What? Is it normal for things to explode? No it is not normal... yes I use the rear battery pack and it worked just fine but then I desoldered the rear pack and soldered a larger one to vbat and gnd and accidentally put one battery in backwards and the part exploded. I did the exact same thing to five other boards and as long as you don't put the battery in backwards they don't explode.

lewisxhe, 2020-07-11:
No, you misunderstood, I mean, have you used the battery holder provided on the back of the board for charging before?  Is it normal?  Or is it your first time using the charging function?

Mr-HaleYa, 2020-07-11:
I said yes I used it, then I removed it.

Mr-HaleYa, 2020-07-16:
@lewisxhe 
any other questions? did your engineer reply?

What part is it and where is it on the schematic and I will replace it.

lewisxhe, 2020-07-16:
[image]

This is a boost IC, you can replace it if you can

nissimzur, 2020-07-17:
This chip boost Vbat 3.6 to 4.2 needed by GSM.
If you remove U11, you can hook to C32 the plus side to external 4.2 V
if you use USB 5V. you can put a diode in series
[image]
 with 5V and get easy 4.2 V.

Mr-HaleYa, 2020-07-17:
@nissimzur 
are you the engineer/developer?

nissimzur, 2020-07-17:
I am. 
look for Dr. Nissim Zur at youtube

Mr-HaleYa, 2020-07-17:
I was if you are the person who built the TTGO SIM7000G?

lewisxhe, 2020-07-20:
This is a stupid approach, and the Modem may not be turned on when using battery power. He is not an engineer of LilyGo.

lewisxhe, 2020-07-20:
The method of connecting the diode in series does not consider the use of battery power supply. When the battery is used, the Modem may not be opened smoothly

lewisxhe, 2020-07-20:
The engineer just told me to use the negative battery on the back of the board instead of GND, which has battery over-discharge, overload, and short-circuit protection.

Mr-HaleYa, 2020-07-20:
That's not an option, its mounted on a PCB
[image]

Mr-HaleYa, 2020-07-20:
@lewisxhe 
Ok, I rebuilt my PCB to have spring contacts to connect to the Battery pins on the back of the TTGO when it is mounted on a PCB. This should add the following benefits 
- Backward battery protection
- overcharge Protection
- Over-Discharge protection
- overcurrent protection
- on/off switch

I think I will still add an STM1061N27WX6F To monitor the battery and force the MCU off (RST LOW) until the voltage is greater than 2.7v so it does not give the bootloader problem

closed 2020-07-29 by Mr-HaleYa