Xinyuan-LilyGO/LilyGO-T-SIM7000G #41: Power question

Mr-HaleYa, 2020-07-14:
If I hook a battery to Vbat and GND so that is being powered, Do I need to remove the battery before plugging into USB? I have plugged into USB with battery in the socket on the back but never with and external battery. I don't want to fry another board...

lewisxhe, 2020-07-15:
Yes, I just did the test
[image]

Mr-HaleYa, 2020-07-15:
So I can have battery connected as you do and plug-in USB?

lewisxhe, 2020-07-16:
[image]
Yes, I tried again, no problem

closed 2020-07-16 by Mr-HaleYa