Xinyuan-LilyGO/LilyGO-T-SIM7000G #43: GPS not Working (20200415)

Mr-HaleYa, 2020-07-17:
I have been having this problem for a long time but I wanted to get #16  fixed first.

I use the TinyGSM example sketch and it never gives me the location it just keeps writing `getGPS [timexxx]` in the serial. 

``` cpp
    // Set SIM7000G GPIO4 HIGH ,Open GPS power
    // CMD:AT+SGPIO=0,4,1,1
    // Only in version 20200415 is there a function to control GPS power
    modem.sendAT("+SGPIO=0,4,1,1");

    modem.enableGPS();
    float lat,  lon;
    while (1) {
        if (modem.getGPS(&lat, &lon)) {
            Serial.printf("lat:%f lon:%f\n", lat, lon);
            break;
        } else {
            Serial.print("getGPS ");
            Serial.println(millis());
        }
        delay(2000);
    }
    modem.disableGPS();

    // Set SIM7000G GPIO4 HIGH ,Close GPS power
    // CMD:AT+SGPIO=0,4,1,0
    // Only in version 20200415 is there a function to control GPS power
    modem.sendAT("+SGPIO=0,4,1,0");
```

nissimzur, 2020-07-17:
Maybe better to US IDF and the delay will be multi-tasking delay that allows other tasks to work parallel. 
Look for IDF sample of vTaskDelay(200 / portTICK_PERIOD_MS);

Mr-HaleYa, 2020-07-17:
What? isn't vTaskDelay a FreeRTOS function and not Arduino IDE?

nissimzur, 2020-07-17:
If you like to write a real-life application. Use expressive IDE https://github.com/espressif/esp-idf Not Arduino IDE. 
one line and you have multitasking apps, that all the tasks are not blocked. 
In all my application to ESP32, I have more than 10 procedures that runs in parallel. 
For Example: 
Let say I like to reset esp32 board ever 1 Hour
void reset_every_1H(void *pvParameter)
{
        vTaskDelay(3600000 / portTICK_PERIOD_MS);   //delay 1H      
        
        char str[10]="esp_restart, 3600000 ms pass";
        printf("\n\n\n\n\n%s [__FUNCTION__: %s lINE:%d] \n\n\n\n\n", str, __FUNCTION__, __LINE__); 
        esp_restart();
}
.....
call in App_main()
	xTaskCreate(&reset_every_1H, "reset_every_1H", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
in the normal procedure you put at end of the procedure
     vTaskDelay(2000 / portTICK_PERIOD_MS); // so it will wait 2 sec before loop again

and make procedure in an endless loop. as while (1)

Mr-HaleYa, 2020-07-17:
Project guidelines are Arduino IDE only

lewisxhe, 2020-07-17:
If you can say this, you don’t know enough about Arduino.  Arduino is more suitable for rapid development. The APIs that IDF can use are also available in Arduino. The only drawback is that it is not the latest.

lewisxhe, 2020-07-17:
You said you have two boards, and neither board can be positioned?@Mr-HaleYa

Mr-HaleYa, 2020-07-17:
I have 10 boards and none of them can be positioned. The antennas are plugged in and they're positioned in a window hopefully have better service yet they never get a position and I've left them running for days

lewisxhe, 2020-07-18:
Wow ? 10 will not work, I think it is impossible. At present, the SIM7000G board is shipped after testing by me. Can you measure the voltage at this point to see if it is 1.8V, and visually observe that there is no obvious false soldering in the following parts of the component.
[image]

Mr-HaleYa, 2020-07-18:
at what point? there are a lot of components right there

Mr-HaleYa, 2020-07-23:
I have 1.8 volts between that point and ground. Do you have to connect to a mobile network first? I use to connect to a network than test it but it made no difference. 

First I get 
```
AT+CGNSINF

+CGNSINF: 0,,,,,,,,,,,,,,,,,,,,

OK
```

then I get this in my serial 
```
AT+CGNSINF

+CGNSINF: 1,0,,,,,,,0,,,,,,1,,,,34,,

OK
```

then after a while, I get 
```
AT+CGNSINF

+CGNSINF: 1,0,,,,,,,0,,,,,,,,,,,,

OK
```

lewisxhe, 2020-07-23:
I have encountered this situation. This is a problem with the antenna connection. Please make sure that the connection is reliable.

Mr-HaleYa, 2020-07-23:
Like I said I have many boards that I have tried this with and always the same result... I have taken off the antenna and placed it back many times trying to make it better. Do you think it is bad signal? I am on the bottom floor in a semi-basement (half under half above ground) but I am right next to the window and put the antenna in the window when testing.

lewisxhe, 2020-07-23:
I suggest you try it outdoors.

Mr-HaleYa, 2020-07-23:
I was just able to get my old laptop (Chromebook 🤮 ) to work with serial monitor so now I can!!! I will do a LOT of testing in the morning. (4am here)

Mr-HaleYa, 2020-07-23:
I took it up to my top floor and GPS connected multiple times in around ~20 seconds so it was bad service in the lower level that caused it. I still won't connect to LTE tho so that was not fixed by moving. 

Im going t close. if I start to have problems again I will reopen

closed 2020-07-23 by Mr-HaleYa

procopio, 2020-07-30:
Hi, I bought 25x LilyGO-T-SIM7000G Version 20200415, but before that I bought 5x Version 20191227. 
I made a lot of tests with the old Version 20191227 and everything was working perfect, so I decided to buy 25 of new version to make production tests before buy more 150 boards. 

Unfortunately in this new version is very hard to getting the GPS signal, including in the middle of the road. 

My program is very simple, it does:
- set the baudrate to 115200
- wait for +CPIN: READY
AT+CFUN=1
AT+CGDCONT=1,"IP","orangeworld"
AT+CIPSHUT
AT+CGATT=1
AT+SAPBR=3,1,"CONTYPE","GPRS"
AT+SAPBR=3,1,"APN","orangeworld"
AT+CSTT="orangeworld"
AT+SAPBR=1,1
AT+CIICR
AT+CGNSPWR=1
AT+CGNSMOD=1,1,1,1

After all this sequence correct, waiting for the OKs, I have a loop every second calling AT+CGNSINF to get gps data.

In the old version it's very fast to start to receive valid data, but in this new version the gps data never is valid. Or just a few time I got some valid data.

Can you please help me with this problem? 

Tks

Gabriel-Gardin, 2022-09-25:
Hello, I am currently having problems with GPS too, sometimes it works, and sometimes it doesn't. And it doesn't seem to have a clear pattern of when it works, and when it doesnt.

lewisxhe, 2022-09-26:
Use `modem.enableGPS()` to check whether GPS is enabled. SIM7000 must drive the GPIO of the modem to enable the GPS circuit. Use the following code to turn on the GPS power supply
````
modem.sendAT("+SGPIO=0,4,1,0");
         if (modem.waitResponse(10000L) != 1) {
             Serial.println("SGPIO=0,4,1,0 false ");
         }
````
@Gabriel-Gardin

Gabriel-Gardin, 2022-09-26:
Modem power supply is enabled... Here you can see all the commands with a logic analyzer:

[image]

lewisxhe, 2022-09-26:
emmm. Please try the following commands:

````
AT+CGNSCFG=2
AT+CGNSTST=1

````
Check if NMEA messages are output to AT Port
@Gabriel-Gardin

Gabriel-Gardin, 2022-09-26:
Still nothing...
When you run AT+CGNSTST=1 it alway output data right? Even when GPS has no fix?

lewisxhe, 2022-09-26:
Right even if the GPS is not fixed, it should always output unfixed NMEA sentences.

Gabriel-Gardin, 2022-09-26:
Sure, thanks for helping. I will receive more boards today or tomorrow, I will try those and see if it's work. I am also thinking of updating the SIM7000G Firmware... I will keep you updated...

Gabriel-Gardin, 2022-09-26:
Have tested with a new board and with a recently updated firmware, same behavior, no data after AT+CGNSTST=1. 
[image]

lewisxhe, 2022-09-27:
Try the following command and it should output correctly

``
AT+SGPIO=0,4,1,0
OK
AT+CGNSTST=1
OK
AT+CGNSCOLD
OK
```

![image](https://user-images.githubusercontent.com/22990954/192408832-a4df0f7c-59a6-4cdc-acf0-4871506d28df.png)

Gabriel-Gardin, 2022-09-27:
Did not work, no nmea sentences.

Gabriel-Gardin, 2022-09-28:
Thanks!!! That was the problem, 
CGNSNMEA was set to 512
I fixed with AT+CGNSNMEA=237279