Xinyuan-LilyGO/LilyGO-T-SIM7000G #45: I can help wil LilyGo SIM7000G development

nissimzur, 2020-07-17:
You can call my WhatsApp for chatting about real-time firmware needed as with 
"LilYGo SIM7000G"
my phone +972544500635 call me with WhatsApp 
I offer you to move to ESP32 IDF ASAP. Andrino may mess your code. 
Use xTaskCreate as much as passable. 
demo:
    xTaskCreate(&reset_every_1H, "reset_every_1H", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
    vTaskDelay(2000 / portTICK_PERIOD_MS);

-Dr. Nissim Zur

Mr-HaleYa, 2020-07-17:
> I offer you to move to ESP32 IDF ASAP. Andrino may mess your code.

This is made to be an Arduino board running Arduino code

nissimzur, 2020-07-17:
I know LiliyGo use Arduino.
But demo code can be copy to Espressif IDE. 
If you work for fan, Arduino is OK. 
If you like to work professionally... there is no other way. in Espressif IDE you got the newer update API to get the most needed task. 
For example, can you set your esp32 to wifi transmit to other esp32 to 1Km with Arduino?

Mr-HaleYa, 2020-07-17:
As of right now, there is no need to use Espressif IDE, The Arduino IDE has worked for this for many years. It's a lot of extra work and completely rebuilding the code for no reason.

If somebody wants to take there time and write all the documentation and example scripts than they can do that but for now Arduino IDE will continue to be used in farther development. 

 also, This is made for cellular data transmission, not 1km wifi... 🙃

lewisxhe, 2020-07-17:
You should contact LilyGo, not here.

closed 2020-07-17 by lewisxhe