Xinyuan-LilyGO/LilyGO-T-SIM7000G #46: philo-create | cant get AT to work

Mr-HaleYa, 2020-07-21:
creating this for @philo-create so we aren't in #38

Mr-HaleYa, 2020-07-21:
if you can get the reset sketch to pass then try and run the allFunctions sketch with AT debugging enabled and share the serial output. 

tip: select text in serial monitor then use CTRL+C to copy then CTRL+V to paste

philo-create, 2020-07-21:
Hey so i cannot get the  reset sketch to  pass .below is the  output  of  what i get  when  i  run the  all function  sketch
 `
Wait...
Initializing modem...
AT+CFUN=0
Failed to restart modem, attempting to continue without restarting
AT+GMM
Modem Name: SIMCom SIM7000
ATI
Modem Info: 
AT+CPIN?
AT+CPIN?
AT+CPIN?
AT+CPIN?
AT+CPIN?
AT+CNMP=38
AT+CMNB=1
Waiting for network...
AT+CEREG?
AT+CGREG?
AT+CEREG?
AT+CGREG?
AT+CEREG?
AT+CGREG?
AT+CEREG?
AT+CGREG?
AT+CEREG?
`

Mr-HaleYa, 2020-07-21:
ya, that means completely no AT response... umm, unplug and make sure everything is disconnected and try again. Mine can get locked and doing that can help. Are the Net Status LEDs on? two little red ones at the bottom. if so what does it look like and is one blinking and if so how fast per second

philo-create, 2020-07-21:
the two little LEDs close to the battery switch are  both on .the green led is  on  since i am running the  board  from usb  power and  the other led close to it is blinking 3 times per second while the led by  the  gps  antenna is blinking once per second

Mr-HaleYa, 2020-07-22:
hmmmmm.... what is the baud rate of the sketch you where previously using? and what rate were you setting the modem? can you try the ATdebug sketch and just see if you can get it to reply with AT it should say OK if so then try the allFunctions sketch with AT debugging enabled again.

philo-create, 2020-07-22:
The board rate that i was using before it stopped working was 115200...switched  from 115200 to 9600 yesterday when i realised i got more response  from the  board while using 9600.I got the OK reply  while testing  the ATdebug sketch and bellow is what i get when i test the allFunctions sketch  again.

///////////////////////////
SDCard MOUNT FAIL

Wait...
Initializing modem...
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
1⸮AT
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CPIN: READY

OK
AT+GMM

SIMCOM_SIM7000G

OK
Modem Name: SIMCOM SIM7000G
ATI

SIM7000G R1529

OK
Modem Info: SIM7000G R1529
AT+CPIN?

+CPIN: READY

OK
AT+CNMP=38

OK
AT+CMNB=1

OK
Waiting for network...
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

philo-create, 2020-07-22:
so eventually AT+CGREG?  gives  me  an  output  of   3  (+CREG: 0.3'). which  from  my  understanding  means  that the  network  has  failed  to  connect..it  seems  like  right  now  everything  is working  fine  but  i still don't understand  why the  board still doesn't  connect to the  network.

Mr-HaleYa, 2020-07-22:
ok, this is the progress I was looking for 👍 I have an active issue open on the tinyGSM library and I think it has to do with there latest updates. 0,3 means connection denied. 0,2 means searching and that is really good because I have one stuck in 0,0 which is not searching and i can't get it out... If you get it to start working PLEASE let me know. I will keep you up to date on my progress as well.  

https://github.com/vshymanskyy/TinyGSM/issues/421

Mr-HaleYa, 2020-07-22:
maybe add to that saying you have 0,3 problem to so the dev is more likely to try and work on it?...

SRGDamia1, 2020-07-22:
I'm coming over from TinyGSM.  

```+CEREG: 0,2``` means you're searching for the network and not connecting.  If you get that it (usually) means either there is no service, your antenna isn't good enough, or your battery/power supply isn't good enough.  It could also mean there's something up with your SIM or your settings, but that's less likely.

```+CEREG: 0,3``` means that the network explicitly denied your request to register.  Sometimes it's intermittent (the module will search for a while (2) then be denied (3) then search for a bit longer (2) and be accepted (1 or 5).  If you're getting it consistently it's probably an issue with your SIM card or (less likely) your connection (band, carrier, etc) settings.

philo-create, 2020-07-23:
@Mr-HaleYa ,I dont  think the  problem  is  with  TinyGsm ,i tested the code again  using  the  previous  version and  i still have  the  same  results .@SRGDamia1 its also  not  the sim card since  i have  another  TTGO T-Call SIM7000G boards that are  working fine . also tested  the codes  on the   TTGO T-Call SIM800L board and  it works  fine.

Mr-HaleYa, 2020-07-23:
none of the problems are specifically the tinyGSM library, I was just saying I'm having the same issue and i was asking there because I think the person who wrote the code that talks to the sim7000 knows more about what to do than I do. SRGDamia1 is very talented, that's why I asked and mentioned the issue.

@SRGDamia1 
I have multiple SIM cards and all of them used to work and I can put them in the sim800 and they work with 2G. I can scan for networks and it shows them using cops=? I am in a half-basement (half underground half above) but I bought a largeish antenna and I am located right next to the window. CatM was working just fine and 2g worked flawlessly.  It is only recently that nothing is working. I do not own a laptop so I'm not able to view serial anywhere except where I sit that is why I have not been able to test anywhere else. Is there a way to test signal quality without explicitly connecting to the network? 

Last time I was able to successfully connect to catm1 this was the serial log so I had Ok service
```
[55593] Operator: AT&T Hologram
[55639] Local IP: 10.117.120.155
[55677] Signal quality: 19

+CPSI: LTE CAT-M1,Online,310-410,0x9709,111159055,393,EUTRAN-BAND12,5110,3,3,-14,-111,-79,10
```

this is what I bought, I also try with my other antennas that all use to work.
[image]

Mr-HaleYa, 2020-07-26:
@philo-create Any luck lately?

lewisxhe, 2020-07-29:
I can't test CAT-M1 in my country. I can't help you, but it keeps showing registration. This must be the reason for the bad signal.

lewisxhe, 2020-07-29:
@philo-create If there is no artificial change of the baud rate, it defaults to 115200. I am surprised that I have encountered two people with this problem.

Mr-HaleYa, 2020-07-29:
@lewisxhe I have never used 115200 baud before... mine instantly responses to 9600 after the first flash. I always use 9600 because it is a slower baud and gives a bit more room for replies, to quick of communication can cause the Sim7000 to run commands out of order.

philo-create, 2020-07-29:
@Mr-HaleYa,I still haven't  had any luck yet.

Mr-HaleYa, 2020-07-29:
@philo-create So are AT commands working now? it just wont connect to service? If so I think i can fix that (maybe)

philo-create, 2020-07-29:
Yes  AT  commands  work and  the  problem is that  it  wont  connect  to any service.

Mr-HaleYa, 2020-07-29:
what provider are you using for service and what is the APN? can you upload the ATdebug sketch (updated version from github) I would like to test some commands

Mr-HaleYa, 2020-07-29:
once you have the ATdebug sketch running correctly try this command and copy-paste the output here 
AT+CGDCONT?

philo-create, 2020-07-29:
output:AT+CGDCONT?

+CGDCONT: 1,"IP","internet","0.0.0.0",0,0,0,0
+CGDCONT: 13,"IP","internet","0.0.0.0",0,0,0,0

OK

Mr-HaleYa, 2020-07-29:
What is the APN you are using?

philo-create, 2020-07-29:
the APN  is "internet"

Mr-HaleYa, 2020-07-29:
hmmmmm then it should be working.... try
 AT+CREG? 

AT+CGREG?

AT+CEREG?

philo-create, 2020-07-30:
AT+CREG?  

+CREG: 0,0

OK
AT+CGREG?

+CGREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK

Mr-HaleYa, 2020-07-30:
very interesting
ok change all of those with these

AT+CREG=2

AT+CGREG=2

AT+CEREG=2

then run 

AT&W
 (this saves the last commands) 

then set it to verbose errors (errors with a description instead of just ERROR)

AT+CMEE=2

after that run these and tell me the outputs

AT+CFUN?

AT+CPIN?

AT+COPS=? 
(this will take a minute to run so be patient)

AT+CPSI?

philo-create, 2020-07-30:
AT+CFUN?

+CFUN: 1

OK
AT+CPIN?

+CPIN: READY

OK
AT+COPS=?

+COPS: (1,"VodaCom-SA","Vodacom","65501",0),(3,"MTN-SA","MTN","65510",0),,(0,1,2,3,4),(0,1,2)

OK

+CREG: 2

+CEREG: 2
AT+CPSI?

+CPSI: NO SERVICE,Online

OK

+CREG: 0

+CEREG: 0

Mr-HaleYa, 2020-07-30:
who are you trying to connect to? Vodacom? because that's who it picks up when scanning for carriers

philo-create, 2020-07-30:
For control i have been testing all the  service providers which also includes vodacom.

Mr-HaleYa, 2020-08-03:
I am at a loss. I made a Pull Request of an improved AllFunctions.ino that adds PDP auto-correction. That was my main problem. when you change APN in the code it was not (and still does not unless you use my new code) change the PDP all of the time. This would make the connection very unreliable and most times impossible.

philo-create, 2020-08-03:
are  your  boards  also  still  not working?

Mr-HaleYa, 2020-08-03:
no, I was able to fix all 5 of them that where not working. That's why I added the code that fixed them to the AllFunctions.ino

Mr-HaleYa, 2020-08-15:
@philo-create 
Have you done any more testing or got them to work? This issue is going stale

Mr-HaleYa, 2020-08-17:
@philo-create 
I'm going to close this for now. if you need more help leave a reply asking and I will reopen it.

closed 2020-08-17 by Mr-HaleYa