Xinyuan-LilyGO/LilyGO-T-SIM7000G #47: ro85ac | Solar questions

Mr-HaleYa, 2020-07-23:
making this for the user so it's not in a closed issue. #26

Mr-HaleYa, 2020-07-23:
I was meaning how many did you buy. Did you only buy 1?

What are the sun conditions? does the sun directly hit the panel? 
the angle the sun hits it can have a huge impact on performance as well as any shade so you want the sun to hit it directly.

ro85ac, 2020-07-23:
@Mr-HaleYa the sun conditions are not the best these days (cloudy). Now for e.g. the panel power is 3.6V. Do you know if this can reach 6V?

ro85ac, 2020-07-23:
@Mr-HaleYa But even if the wheather is not so good, the battery should charge a bit or at least not to drop. I would like to use this board in autumn, so in 3 rainy days the battery will be dead.

Mr-HaleYa, 2020-07-23:
yes, I understand, Have you check the connection to make sure it is correct? it appears that it did charge a little the one day, correct? If the solar panel is backward a yellow led should be on. if it is correctly connected it should have a green and/or red LED on
red = charging, green = full, red+green blinking means error.

can you take a picture of the board while connected to the panel

ro85ac, 2020-07-23:
@Mr-HaleYa The led is red. [image]

Mr-HaleYa, 2020-07-23:
my most current hypothesis is that your panels simply can't supply enough current to charge the batteries. because of ohms law when the battery starts charging the resistance increases making the voltage and current drop on the solar panel. You can do 2 things to fix your problem if this is the case
 1 - buy a bigger solar panel with higher current, the next size up looked a lot better with double the current.
 2 - take 2 or 3 of your small panels and hook them in parallel NOT SERIES! so that the current is 2x or 3x that of 1 making the voltage more stable. 

OR you can be like me and buy a 3 dollar MPPT and a higher voltage solar panel like 12v and a battery pack of 3-4 cells so it would last a LOT longer and charge way faster.

ro85ac, 2020-07-23:
@Mr-HaleYa 
Option 1. What should be the correct Volts and Ampers for the panel?

Option 2. Can you please send me the links where you bought the pieces?

Mr-HaleYa, 2020-07-23:
well, the voltage it shows when in the sun and the voltage it shows when connected to the battery will be very different. like my 12v panel shows 27v in the sun with nothing connected because there is no charging resistant so the current is High. but when connected to a battery charger its only 6volts on the panel because it's charging at 3amp output (a LOT). 

the best thing to do if you aren't sure is just get a physically larger panel but at the same voltage and it will have a higher Watt value.

You still dint answer if you have more than one panel... I think you should try option one before option 2 unless you want option 2

ro85ac, 2020-07-24:
@Mr-HaleYa 
I only have 1 panel of 6V 1A. So you say to increase the Volt (9V 1.12A ) or the Amper (6V 1.67A) ?

Mr-HaleYa, 2020-07-24:
Looking at the panels some more and doing the math the smaller panels _should_ be able to charge it just fine. the problem may be that the chip on the old board is NOT a solar charge but just a regular charging chip. The new board is an actual solar charging chip. 

increase the amps which will increase the Watts (Voltage x Amps = Watts). So buy a higher Amped same voltage solar panel. for example the 6V 1.67A https://www.aliexpress.com/item/32877917510.html

The charging chip onboard has no protection IC so if you overvolt it it will die... 
That's why I decided to go with an external charger. I also went with an external power pack, This removes the on-off switch tho so If you do decide to go this way you won't be able to turn it off without unless you solder the battery pack to the pins on the back of the board where the battery pack is now or disconnect the pack. 

the solar panel we use https://www.amazon.com/gp/product/B0833NQLKR/ref=ox_sc_act_title_4?smid=AV16HZDTPF1LD&psc=1
(its like way overkill but it charges soooo fast)

the MPPT we use https://www.aliexpress.com/item/32857974617.html?spm=a2g0s.8937460.0.0.718d2e0eStZrGj
(rated for 6-28v in and 3.7v out )
It will charge up to 4.2 then shut off for the day so it doesn't overcharge

let me know what you want to do

ro85ac, 2020-07-24:
@Mr-HaleYa 
For now I will try with a more powerful solar panel and the new board. From I read here: https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G the new board has protection for overloading and over-discharge.

Thanks for help!

Mr-HaleYa, 2020-07-24:
you're welcome. also, the new board should charge a LOT better because its an actual solar charger. 

Would you like me to close this now?

ro85ac, 2020-07-24:
@Mr-HaleYa  yes close it now!

Mr-HaleYa, 2020-07-24:
If you need any more help with this issue please comment and I will reopen 
Pleasure helping you
  \- Hale

closed 2020-07-24 by Mr-HaleYa