Xinyuan-LilyGO/LilyGO-T-SIM7000G #51: Sim7000G and LoRa

rodri16, 2020-08-03:
Hi, do you have any example on how to connect and an example of Sim7000G and Lora Board? I would like to have a system with LoRa and NB-IoT connection according to radio coverage.

Here you say that it is possible to connect LoRa shield
[http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1246&FId=t3:50033:3](url)

Mr-HaleYa, 2020-08-05:
That link goes nowhere...
Are you using an official LoRa board that is so along with there or is a 3rd party board if so give details, and images/links?

slavino, 2020-08-14:
https://www.aliexpress.com/item/4000542688096.html

[image]

0lav, 2020-08-15:
I am going to be using one of these soon with the SIM7000G. I will share my results when its done. 

https://images-na.ssl-images-amazon.com/images/I/41KYtRSkaZL._AC_.jpg

lewisxhe, 2020-08-21:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_Cayenne 
It is an example of applying LORA to TTN

Mr-HaleYa, 2020-09-20:
@rodri16 @lewisxhe 
The solution was given and now they even sell a product for LORA so this should be closed

closed 2021-03-01 by lewisxhe