Xinyuan-LilyGO/LilyGO-T-SIM7000G #53: It's possible to restart SIM7000G?

procopio, 2020-08-15:
After many time (about 24h) running one of my boards stop to respond to AT commands.
Is it possible to restart the SIM7000G?

I already tryed:
`
#define FONA_PWRKEY 4
#define FONA_RST 5

void reset()
{
	digitalWrite(FONA_RST, HIGH);
	delay(10);
	digitalWrite(FONA_RST, LOW);
	delay(100);
	digitalWrite(FONA_RST, HIGH);
}

void powerOn() 
{
	digitalWrite(FONA_PWRKEY, LOW);
	delay(100); // For SIM7000
	digitalWrite(FONA_PWRKEY, HIGH);

	delay(100);
	reset();
}
`

procopio, 2020-08-16:
I also tried to reset the Arduino board using the RESET button but the SIM7000G only come back if I power off the board disconnecting the USB cable and switch off.

Mr-HaleYa, 2020-08-16:
If your board stops responding to AT then you are doing something wrong in the sketch. My company has these installed all across our city for data collection, they have been running for months and all reply to AT commands just fine. Not a single one has gone "bad". 

___

Assuming you just want to REBOOT the sim7000g you can use 

```cpp
#define MODEM_RST     5

void modem_reset() {
  Serial.println("\nModem hardware reset\n");
  pinMode(MODEM_RST, OUTPUT);
  digitalWrite(MODEM_RST, LOW);
  delay(260); //Treset 252ms
  digitalWrite(MODEM_RST, HIGH);
  delay(10000); //Modem takes longer to get ready and reply after this kind of reset vs power on
}
```

This will do the same thing as pressing a REBOOT button on the sim7000g (if it had one).
you can also just disconnect all power (batteries and USB) then plug it back in. 

the button on the board called RST is for the ESP32 to REBOOT

if it is not replying to AT after this you may need to RESET the sim7000g. there is a sketch made to reset the board. 
/examples/Arduino_TinyGSM/[ResetModem](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/ResetModem/ResetModem.ino)
It is not a perfect reset tho because they don't have one for the sim7000 yet so it's using the sim800 code (very very similar) 

___

Another thing, we don't use the FONA library. that is adafruit. it is super old, outdated, and not maintained as it should be. If you are just trying to build something simple, FONA might work. I have done a LOT of development with Sim7000 modules and I also have tried using FONA. I hated it. they use bad methods of "talking" to the sim7000 that can result in a "bricked" device (AT not working). 

So for those reasons I highly recommend using TinyGSM since it is under active development. 

___

If this has fixed your issue CLOSE it, If you need more help, leave a reply 🙂 

\- Hale

procopio, 2020-08-30:
After 72hrs running the boards in the city none has stopped.
This is a good news, but I use this boards inside cars moving into areas without signal.
So it concern me if it will stop or not...
I'm still guessing the problem could be caused by a power consumption peak.
There is some point of the board where I can add a capacitor to hold the power to avoid this kind of situation?
Many thanks for your help!

grichfitz, 2020-09-05:
Hay procopio 
I am also in Spain and not having any issues like this. What carrier are you using for 2G?

procopio, 2020-09-14:
Hi @grichfitz , we are using Orange and you?

Mr-HaleYa, 2020-09-20:
Why do you think that is caused by power consumption peak? You think the power draw spikes or dips and then it locks the sim7000? Adding a capacitor with smooth spikes but I really don't understand your thinking of power spikes

If you want to restart the modem I explained how but you said it didn't work yet you didn't really explain what you meant by that. When they stop working do the little red LEDs keep blinking or do they both shut off or what is the state of the board?

procopio, 2020-09-30:
Found the problem! 
At some point we was sending AT commands without wait the response from another command.
I think this was crashing the sim7000 someway.
Now we learn to send one command per time.
Thanks for you all and hope this help someone.

closed 2020-09-30 by procopio

Mr-HaleYa, 2020-10-04:
> Found the problem!
> At some point we was sending AT commands without wait the response from another command.
> I think this was crashing the sim7000 someway.
> Now we learn to send one command per time.
> Thanks for you all and hope this help someone.

Yes, that can cause crashes. 
I hope you have fixed your problem but if you are still having disconnections I found that mine was having them because it was trying to connect to incorrect bands. I removed all bands except band 12 from catm1 and now it hasn't disconnected once for a week. before it disconnected every 52-85 minutes 

if you are using catm1 or NBIOT then you can see the bands with this AT command `AT+CBANDCFG?` 
and set you own bands with `AT+CBANDCFG=CAT-M,12`  (I use catm1 band 12 only)

procopio, 2020-10-05:
Tks for the tip @Mr-HaleYa! 
My problem with AT command was the AT+CGATT=1 has timeout of 140 seconds to give an error, but in the AT Commands document it is wrong (75 seconds).
After the 75 seconds I was trying to send a new command and the modem stop responding.