Xinyuan-LilyGO/LilyGO-T-SIM7000G #55: Test functionality problem

AndreaCalifano4, 2020-08-19:
hi I'm trying the "AllFunctions" script but the device doesn't seem to connect, the serial minitor keeps showing "waiting for network ...". Also trying with the ATdebug script I can't use the GPS: the command AT + CGPS =? returns ERROR.

Tsaadon, 2020-08-20:
Hey, I have the same issue,  trying the "AllFunctions" script and getting stuck in "waiting for network ..."

lewisxhe, 2020-08-21:
What network are you using? You need to check your card and carrier. If you have a 2G network in your area, you can change modem.setNetworkMode(38); to modem.setNetworkMode(13); to try a 2G connection, you must confirm that there is a signal at your location  

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/2a628f1e29d64a269dca55a0cc9358cc06f1ea7f/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino#L119

AndreaCalifano4, 2020-08-21:
> What network are you using? You need to check your card and carrier. If you have a 2G network in your area, you can change modem.setNetworkMode(38); to modem.setNetworkMode(13); to try a 2G [...]

With 2G It work.
And about GPS what could go wrong? With the command AT+CGPS=? Return ERROR

lewisxhe, 2020-08-25:
Turn off GPS use `AT+CGNSPWR=0` @AndreaCalifano4

Mr-HaleYa, 2020-09-20:
@AndreaCalifano4 @lewisxhe 

The solution was given and one person said it worked and there's been no activity for over 30 days so this should be closed

closed 2021-03-01 by lewisxhe