Xinyuan-LilyGO/LilyGO-T-SIM7000G #56: Ring indicator pin not connected

bindi, 2020-08-21:
Hello,
Will there be a new revision of the 7000 with the RI (Ring indicator, pin 4) pin available? This is a must-have for low power applications. I was disappointed to see it not connected on the 800 version, and yet again disappointed to see it left unused in this board as well.

lewisxhe, 2020-08-21:
SIM7000 does not have the call function. You may buy the earlier version of IP5306 for SIM800. Later AXP192 power supply will have RI indication. After a while, IP5306 will be upgraded and RI indication will be connected to ESP32.

closed 2020-08-21 by lewisxhe

bindi, 2020-08-25:
https://simcom.ee/documents/SIM7000E/SIM7000%20Hardware%20Design_V1.05.pdf
3.3.2 RI and DTR Behavior
Am I misunderstanding the datasheet or why does the sim7000 have the RI pin at all then? By call function do you mean it can not receive or send phone calls? The Ring Indicator can be used for incoming SMS.

robbiet480, 2021-08-16:
@lewisxhe So did the RI pin get connected in the newer version of this board?

lewisxhe, 2021-08-16:
@robbiet480 LilyGo products are no longer maintained by me, please contact LilyGo staff

robbiet480, 2021-08-16:
Thanks for letting me know @lewisxhe!

@LilyGO Can you comment on if the RI pin is connected or not?

LilyGO, 2021-09-02:
> Thanks for letting me know @lewisxhe!
> 
> @LilyGO Can you comment on if the RI pin is connected or not?

> Thanks for letting me know @lewisxhe!
> 
> @LilyGO Can you comment on if the RI pin is connected or not?

The pcie RI is connected to the main control.T-7000g We will connect the RI pin to the master in the new version