Xinyuan-LilyGO/LilyGO-T-SIM7000G #57: ttgo sim7000g does not go to sleep

naturalmechanics, 2020-08-25:
Hello, I am trying to sent the ttgo to sleep, here is my code. 

```
int i = 10;
	while (i) {
		SerialGSM.println("AT+CGNSPWR=0");																				//GNSS Power on
		delay(500);
		if (SerialGSM.available()) {
			String r = SerialGSM.readString();
			Serial.println(r);
			if ( r.indexOf("OK") >= 0 ) break;;
		}
		delay(500);
		i--;
	}																													Serial.println("internal GPS power DOWN");
	
	i = 10;
	while (i) {
		SerialGSM.println("AT+SGPIO=0,4,1,0");														Serial.println("GNSS pin DOWN");
		delay(500);
		if (SerialGSM.available()) {
			String r = SerialGSM.readString();
			Serial.println(r);
			if ( r.indexOf("OK") >= 0 ) break;;
		}
		delay(500);
		i--;
	}
	
	
	//delay(1000);
	
	
	digitalWrite(PIN_DTR, HIGH);																						Serial.println("MODEM DOWN");
	delay(1000);
	
	
	i = 10;
	while (i) {
		SerialGSM.println("AT+CSCLK=1");																				//modem power down
		delay(500);
		if (SerialGSM.available()) {
			String r = SerialGSM.readString();
			Serial.println(r);
			if ( r.indexOf("OK") >= 0 ) break;;
		}
		delay(500);
		
		i--;
	}
	
	
	
	digitalWrite(EXTERNAL_PWR_PIN, LOW);																				Serial.println("External Sensor DOWN");
																														Serial.println("GOING to SLEEP NOW");
    
    
	esp_deep_sleep_start();
```

I Expect this to send the system to draw less than 10 mA current. However, the Modem lights both stay on (the network activity light flashes some times), and i draws more than 40 mA, or sometimes even 200 mA current.

What am I doing wrong?

lewisxhe, 2020-09-08:
Please use modem.poweroff(); directly instead of pulling down EXTERNAL_PWR_PIN

ro85ac, 2020-09-16:
@lewisxhe 
How can I call SIM7000 to poweroff in the LilyGO-SIM7000-Cayenne.ino project ?

lewisxhe, 2020-09-17:
As mentioned above, use modem.poweroff()

ro85ac, 2020-09-17:
@lewisxhe 
there is no "modem" member in LilyGO-SIM7000-Cayenne.ino class

Mr-HaleYa, 2020-09-20:
> there is no "modem" member in LilyGO-SIM7000-Cayenne.ino class

Im not meaning any offence but , you really should go take some courses on Arduino... you would learn a LOT and it doesn't take long.
anyways, just add `TinyGsm modem(gsmSerial);` under `HardwareSerial  gsmSerial(1);` and it will create it

Mr-HaleYa, 2020-09-20:
@naturalmechanics 
>What am I doing wrong?

As for you... 

like Lewis said, "use modem.poweroff();" and don't be pulling the power pin... that's like yanking the cord out on your PC instead of telling it to shutdown...

as for your complaint about the LEDs.. if you are that concerned about power than you can always desolder. quick, efficient, and easy. if you want to go with a software aproch then you should read the AT manual as it explains how [Here]( http://www.microchip.ua/simcom/LTE/SIM7000/SIM7000%20Series_AT%20Command%20Manual_V1.05.pdf)
here's is something I've used before for the netlight
```cpp

void turnOffNetlight() {
  SerialMon.println("Turning off SIM800 Red LED...");
  modem.sendAT("+CNETLIGHT=0");
  modem.waitResponse();
  modem.sendAT("+CSGS=0");
  modem.waitResponse();
  modem.sendAT("&W");
  modem.waitResponse();
}

void turnOnNetlight() {
  SerialMon.println("Turning on SIM800 Red LED...");
  modem.sendAT("+CNETLIGHT=1");
  modem.waitResponse();
  modem.sendAT("+CSGS=1");
  modem.waitResponse();
  modem.sendAT("&W");
  modem.waitResponse();
}
```

another thing... go to your ino file in the arduino IDE and press ctrl + T . now do that to every in file you have ever wrote because what you pasted above looks like a child wrote. Im not trying to be harsh but if you posted that anywhere else in the format you would be torn apart by all the ignorant developers. I made the same mistake once and I almost quit coding because of some of the words the replyer said.

javipelopi, 2021-01-27:
Same issue here, using modem.poweroff() turns off the sim7000 for a couple of seconds and then the lights turn on again.

In the TinyGSM sketch it sais:

  // Try to power-off (modem may decide to restart automatically)
  // To turn off modem completely, please use Reset/Enable pins

How should I use the Reset/Enable pins? Any help?

Thanks in advance!

procopio, 2021-02-11:
I'm using the command AT+CSCLK=1 and controlling the sleep using the DTR pin.

closed 2021-03-01 by lewisxhe