Xinyuan-LilyGO/LilyGO-T-SIM7000G #59: Timed out waiting for packet header

ro85ac, 2020-09-15:
I have the new board and I am getting this annoying issue when I try to upload. I google it and all the answers say to press the boot button, but I see on the board PWR and RST buttons. I pressed each one separately and also both of them in the same time, but nothing helped.

```
esptool.py v2.6
Serial port COM5
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```
 
[image]
[image]

cuplesoft, 2020-09-15:
It looks like wrong port number.

ro85ac, 2020-09-15:
The board has a problem with the TypeC connector. It's moving inside it.

closed 2020-09-15 by ro85ac