Xinyuan-LilyGO/LilyGO-T-SIM7000G #60: Voltage dividers for solar & battery

ROSW6341, 2020-09-16:
What is the tolerance for the voltage divider resistors R62, R63, R64 & R65?  After compensation for the ADC non-linearity the voltages are about 5% low.  I installed and external divider using 1% resistors between 3.3v & ground.  That voltage is read accurately (less than 1% off).

lewisxhe, 2020-09-17:
After I asked, I learned that 5% was used. So the accuracy may be a little bit worse

ROSW6341, 2020-09-17:
I 'trick' to use 5% resistors is to use a 2-pack (both resistors on the same substrate).  They might be off by 5% but will be in the same direction still maintaining a constant divide by 2.  A 4-pack could be used for the two dividers.  Unfortunately either would require a board spin.

closed 2021-03-01 by lewisxhe