Xinyuan-LilyGO/LilyGO-T-SIM7000G #64: Connecting using GSM to Azure MQTT failed

ro85ac, 2020-10-04:
Using WiFi connection in this project all the messages are sent to Azure Cloud. But when using a GSM connection I receive the following error.

21:39:41.971 -> Info: Initializing SNTP

21:39:41.971 -> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
21:39:41.971 -> abort() was called at PC 0x40088091 on core 1
21:39:41.971 -> 
21:39:41.971 -> Backtrace: 0x4008b764:0x3ffb1d90 0x4008b995:0x3ffb1db0 0x40088091:0x3ffb1dd0 0x4010677a:0x3ffb1e10 0x40106922:0x3ffb1e30 0x400f8d74:0x3ffb1e50 0x400f8ddd:0x3ffb1e70 0x400ffc91:0x3ffb1e90 0x400ffcb6:0x3ffb1eb0 0x400f8c0b:0x3ffb1ed0 0x400d3af5:0x3ffb1ef0 0x400d3b56:0x3ffb1f10 0x400d2c12:0x3ffb1f30 0x400d2819:0x3ffb1f70 0x400df4ab:0x3ffb1fb0 0x400883a5:0x3ffb1fd0

[SIM7000G_Azure_MQTT.txt](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/5324092/SIM7000G_Azure_MQTT.txt)

closed 2020-10-06 by ro85ac

ejri, 2021-03-21:
@ro85ac were you able to find a solution to this? 

I have the same issue when connected over GSM. Board connects perfectly fine with Azure over wifi, but doesn't connect over GSM. 

With Azure the last thing I see is:

> Modem Connected to Rogers' LTE (channel--> 38) CAT-M (preferred network). TLE CAT-M OK
>  > IoT Hub
> Info: Initializing SNTP

```
// This code: connecting 5 accelerometers to esp32. The processed data is sent to IoT Hub.
// IoT Hub connection, and device provisioning, etc. :

// SimpleMQTT_esp32_azure_iothub.ino
// downloads/ubc_5G/sensor_networks_azure/

// IoT Hub has to be Standard Tier, Basic Tier creates a lot of issues

// Azure ESP32 IoT DevKit Get Started
// https://docs.microsoft.com/en-us/samples/azure-samples/esp32-iot-devkit-get-started/sample/

/**
   A simple Azure IoT example for sending telemetry to Iot Hub.

  original code from:
  https://github.com/VSChina/ESP32_AzureIoT_Arduino
  se
*/

// TCA9548A I2C Switch between:
// adafruit LSM9DS1 at address, on TCA9548's channel 0
// GY521-MPU 6050 at address, on TCA9548's channel 1
// TCA9548 module I2C address:  0x70
//

//======================================================
//I2C Mutiplexing using TCA9548A at I2C address 0x70
//======================================================

// GY-521 / mpu6050 on esp32
// connections: 5v, GND, SDA, SCL, on sensor to 5v, GND, SDA, SCL on esp32 respectively

#include "Esp32MQTTClient.h"

/*
///////////////////////////MQTT device to cloud initial message setup///////
#include <ArduinoMqttClient.h>
#include <ArduinoBearSSL.h>
///////////////////////////MQTT device to cloud initial message setup///////
*/

//////////////////////////// Accelerometer ////////////////////
// Include Wire Library for I2C
#include <Wire.h>
// Amazing MPU6050 library by rfetick.
#include <MPU6050_light.h>
MPU6050 mpu(Wire);

const int MPU = 0x68;
int16_t roll1, pitch1, yaw1, roll2, pitch2, yaw2, roll3, pitch3, yaw3, roll4, pitch4, yaw4, roll5, pitch5, yaw5;
//////////////////////////// Accelerometer ////////////////////

void TCA9548A(uint8_t bus)
{
  Wire.beginTransmission(0x70);
  Wire.write(1 << bus);
  Wire.endTransmission();
}

[... 623 lines cut ...]
    // failed, retry
    Serial.print(".");
    Serial.println(MqttClient.connectError());
    delay(5000);
  }
  Serial.println();

  Serial.println("You're connected to the MQTT broker");
  Serial.println();

  // subscribe to a topic
  MqttClient.subscribe("devices/" + deviceId + "/messages/devicebound/#");
}

void publishMessage() {
  Serial.println("Publishing message");

  // send message, the Print interface can be used to set the message contents
  MqttClient.beginMessage("devices/" + deviceId + "/messages/events/");
  MqttClient.print("hello ");
  MqttClient.print(millis());
  MqttClient.endMessage();
}
*/
```

ro85ac, 2021-04-01:
> @ro85ac were you able to find a solution to this?
> 
> I have the same issue when connected over GSM. Board connects perfectly fine with Azure over wifi, but doesn't connect over GSM. The cayanne example works over GSM though.
> 
> With Azure the last thing I see is:
> 
> > Modem Connected to Rogers' LTE (channel--> 38) CAT-M (preferred network). TLE CAT-M OK
> > > IoT Hub
> > > Info: Initializing SNTP
> 
> ```
> // This code: connecting 5 accelerometers to esp32. The processed data is sent to IoT Hub.
> // IoT Hub connection, and device provisioning, etc. :
> 
> // SimpleMQTT_esp32_azure_iothub.ino
> // downloads/ubc_5G/sensor_networks_azure/
> 
> // IoT Hub has to be Standard Tier, Basic Tier creates a lot of issues
> 
> // Azure ESP32 IoT DevKit Get Started
> // https://docs.microsoft.com/en-us/samples/azure-samples/esp32-iot-devkit-get-started/sample/
> 
> /**
>    A simple Azure IoT example for sending telemetry to Iot Hub.
> 
>   original code from:
>   https://github.com/VSChina/ESP32_AzureIoT_Arduino
>   se
> */
> 
> // TCA9548A I2C Switch between:
> // adafruit LSM9DS1 at address, on TCA9548's channel 0
> // GY521-MPU 6050 at address, on TCA9548's channel 1
> // TCA9548 module I2C address:  0x70
> //
> 
> //======================================================
> //I2C Mutiplexing using TCA9548A at I2C address 0x70
> //======================================================
> 
> // GY-521 / mpu6050 on esp32
> // connections: 5v, GND, SDA, SCL, on sensor to 5v, GND, SDA, SCL on esp32 respectively
> 
> #include "Esp32MQTTClient.h"
> 
> /*
> ///////////////////////////MQTT device to cloud initial message setup///////
> #include <ArduinoMqttClient.h>
> #include <ArduinoBearSSL.h>
> ///////////////////////////MQTT device to cloud initial message setup///////
> */
> 
> //////////////////////////// Accelerometer ////////////////////
> // Include Wire Library for I2C
> #include <Wire.h>
> // Amazing MPU6050 library by rfetick.
> #include <MPU6050_light.h>
> MPU6050 mpu(Wire);
> 
> const int MPU = 0x68;
> int16_t roll1, pitch1, yaw1, roll2, pitch2, yaw2, roll3, pitch3, yaw3, roll4, pitch4, yaw4, roll5, pitch5, yaw5;
> //////////////////////////// Accelerometer ////////////////////
> 
> void TCA9548A(uint8_t bus)
> {
>   Wire.beginTransmission(0x70);
>   Wire.write(1 << bus);
>   Wire.endTransmission();
> }
> 
[... 640 lines cut ...]
> void publishMessage() {
>   Serial.println("Publishing message");
> 
>   // send message, the Print interface can be used to set the message contents
>   MqttClient.beginMessage("devices/" + deviceId + "/messages/events/");
>   MqttClient.print("hello ");
>   MqttClient.print(millis());
>   MqttClient.endMessage();
> }
> */
> ```

@ejri 
1. I don't have a solution yet.
2. The Azure SDK (Esp32MQTTClient) calls another time library  (I don't remember the name now), but it works only with WiFi connection.
3. Using the GSM connection, you need to call the Azure API , something like this:
 // Set the client id used for MQTT as the device id
  mqttClient.setId(deviceId);
  // Set the username to "<broker>/<device id>/api-version=2018-06-30" and empty password
  String username = "";
  // username += broker;
  // username += "/";
  // username += deviceId;
  username += "/api-version=2018-06-30";
  mqttClient.setUsernamePassword(username, "");

ejri, 2021-04-01:
@ro85ac thank you for your reply. I was able to get it working. Check the examples section in this repo, I uploaded an Azure example there. 

Basically, esp32 has a number of features, including the internal clock, that get initiated only after initiating the wifi library...

ro85ac, 2021-04-01:
@ejri 

But if there is no wifi connection does it work?

Which repo? Can you share a link?

Thanks?

ejri, 2021-04-01:
@ro85ac 
Yeah, you just need to initiate the <wifi.h> library, by calling WiFi.begin(); 

You don't have to have it connect to wifi. You just initiate it, with a blank ssid and password. 

[Link> 
](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino)