Xinyuan-LilyGO/LilyGO-T-SIM7000G #65: LILYGO® TTGO T-SIM7000G - Unexpected Shutdown

procopio, 2020-10-08:
Hello,

We have many boards, sometimes, some boards (no pattern) just "shutdown" when running on battery after some minutes.

When it's shutdown all Leds as off, but there are some strange voltage measurements on the board: 
3v3 - GND = 0.7v
VBAT - GND = 0.9v
TX - GND = 0.7

v3v3 - battery negative pole = 3.8v
VBAT- battery negative pole = 4.0v
TX - battery negative pole = 3.8v

The battery is ok and fully charged.
When we push the switch OFF then back to ON, everything is back to normal.
3v3 - GND = 3.3v
VBAT - GND = 4.0v
TX - GND = 3.3v

Also if I plug to USB it comes back to normal too.

Please, anyone have any idea what could be happening?

Thank you all!

MilkoKrastev, 2020-11-17:
I experience the same problem with two boards. They work great if plugged to USB, however, they randomly shutdown when going into deep sleep on battery. 

My program keeps the ULP running for 2 minutes, then wakes up to send some measurements, then goes back to deep sleep. The battery is fully charged, those shutdowns happen completely at random and only when powered by battery. The board could be running fine for 40 minutes, and then shutdown unexpectedly. It could shutdown just after 2 minutes. 

That behavior is a deal breaker for otherwise great board. Is there any GPIO pin that could be stopping the board? Any ideas?

procopio, 2021-01-18:
Hello, after trying many things, I think I solved the problem soldering a cable from battery negative socket to GND in the board. 
I'm not an electronic engineer so I can explain why is that, but looks like it's fixed the random shutdowns. 

Can this cable damage something in the board?

@MilkoKrastev can you please try to see if this works for you too before I close the issue.

Best wishes!

nepat, 2021-01-30:
In my experience the board has worked best when the battery is connected directly to the VBAT and GND pins. The battery of course has to be a 3.7V lithium type. That way the power switch on the board and its questionable logic is bypassed.

procopio, 2021-02-11:
After many tests I can confirm, connecting GND to the negative pole of the battery solved the random shutdowns.

closed 2021-02-11 by procopio

fgnievinski, 2024-01-11:
probably related:
"Short S1 and S2 pins pins each time the battery is reconnected"
[image]