Xinyuan-LilyGO/LilyGO-T-SIM7000G #70: Failed Register to Network in Board 20200415 Version, but Successfully in 20191227 Version

nazmibojan, 2020-10-24:
Hi,

I'm Nazmi and have been purchased two different version of LilyGO-T-SIM7000G board. In 20191227 version, my SIM card can easily registered to network after initialize. It was indicated by NETLIGHT status that was blinking every 3 seconds. But, in 20200415 version, my SIM Card could not register to network. NETLIGHT status always blinkiing every 800ms. If I run AllModemTest example I always got this:

[image]

I have set networkMode to 13, which is GSM only. My SIM card operator use 900MHz band for their 2G ntwork. Please give any insight for solving my problem. Thank you.

Regards,
Nazmi

jupiter8888, 2020-10-31:
One thing you can try is  writng the apn manually with the AT commant sketch. the command is: AT+CGDCONT=1,"IP","YOURAPN"
do that with each PDP context, i noticed there is context 13 active also, so it will be:  AT+CGDCONT=13,"IP","YOURAPN"
after check with AT+CGDCONT?

lewisxhe, 2020-12-15:
You can check whether the two boards use the same options
AT+CNMP=? Return to current mode
AT+CNMP=2 set to automatic mode

nazmibojan, 2020-12-15:
Hi,

Thanks All for your sugestion. But the problem is regulation in my country. Some new regulations have been launched in September about IMEI validation. I bought LilyGo from China without register its IMEI to related minister information system. So, it made my LilyGo couldn't register to LTE network.

Regards,
Nazmi

closed 2020-12-15 by nazmibojan