Xinyuan-LilyGO/LilyGO-T-SIM7000G #73: Randomly failing to register to the network

Elmi77, 2020-12-10:
I have a problem with the SIM7000G registering to the network. Randomly it works, but in 90% of the time it fails.

I'm using code which is quite similar to your example here. No pin is set, the SIM-card is working (verified this with an other device). When - before trying to register to the network - I check the signal quality via getSignalQuality() and it is at 99, the following call to waitForNetwork() first needs a very long time and then fails.

But sometimes it works: then getSignalQuality() returns values around 27 and waiting for the network needs only a few seconds. I could not find out when this happens. Sometimes it helps to vary the parameters for setPreferredMode() and setNetworkMode(), sometimes this has no influence,

I have both versions of the LilyGo SIM700G board and both show the same behaviour. I also tried three different antennae, the result is still the same, I can't get the board working properly.

EmanueleFeola, 2020-12-11:
I have a very similar issue: sometimes it connects to network, sometimes it does not.
Yesterday I tried to reset the modem with the ResetModem.ino script and it failed too.
Then I tried this morning and it was working again (without changing anything)
I'm testing with:
- LilyGO-T-SIM7000G 20-4-15
- hologram SIM card
- power: 18650 battery + usb-c cable
This is the log that I hope is significant since it show that often the esp32 goes bananas
[log.txt](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/files/5678547/log.txt)

Elmi77, 2020-12-14:
Any feedback on this @lewisxhe ?

lewisxhe, 2020-12-15:
Please use AT Debug to view the current network mode:

AT+CNMP=?   Return to current mode
AT+CNMP=2   set to automatic mode

AT+CMNB=?   Check whether it is CAT-M or NB-IoT
AT+CMNB=3   Select CAT-M and NB-IoT

If you use the automatic mode, its network access speed will be relatively slow, you can wait patiently.

If your country still has 2G network, then you can try to use
AT+CNMP=38

Elmi77, 2020-12-15:
@lewisxhe  OK, the problem is somewhat different: no matter what modes I use, it works only _sometimes_. So when it is working, it works with network mode 51 and preferred mode 3. So it is verified the SIM is fine and the network/preferred modes are correct.

But that works only randomly, when it fails, it also fails with these settings. What I noticed: In case of a failure, the signal quality (call to modem.getSignalQuality()) is 99 and the GSM time (call to modem.getGSMDateTime(DATE_TIME)) is somewhere at 0 o'clock.

As getGSMDateTime() can be called before setting network or preferred mode, the reason for the failure seems to be at some earlier stage of initialisation. So playing around with these modes will not help.

Elmi77, 2020-12-16:
@lewisxhe to clarify this, this is my initialisation sequence (pseudocode, based on what is shown in your examples):

`   pinMode(PWR_PIN, OUTPUT);
   digitalWrite(PWR_PIN,HIGH);
   delay(600);
   digitalWrite(PWR_PIN,LOW);
   delay(10000);
   modem.init()
   modem.sendAT("+SGPIO=0,4,1,1");
   delay(600);
   modem.enableGPS()
   modem.getGSMDateTime(DATE_TIME);
   modem.setNetworkMode(2);
   modem.setPreferredMode(3);
   modem.getSignalQuality();
   modem.waitForNetwork()
`

When it fails, getGSMDateTime() returns an invalid time, getSignalQuality() returns 99 and waitForNetwork() is never successful.

When it works (with the same SIM-card and settings as before), getGSMDateTime() returns the current time, getSignalQuality() returns something in range 25..27 and waitForNetwork() is successful after a few seconds.

This lets me think there is something wrong or missing with the initialisation of the SIM7000G - or it is a design problem with the hardware at all.

Elmi77, 2020-12-17:
@lewisxhe so I did some more tests:

- using the example here out of your GIT repository
- using two different boards
- using three different SIMs
- using three different antennae
- playing around with the nework and preferred modes with each

Result: the board always behave the same. So there is either some initialisation missing which is not dosumented anywhere or there is a design problem with the boards in general. For me it looks like the HF-part/mobile network antenna part is wrong. Or in other words: this company is selling unusable crap which can't work as specified.

lewisxhe, 2021-01-13:
Sorry, I did not see your comment. If your board does not work properly, please contact customer service for processing. This test code is not only tested by me, but also by many users. Maybe your hardware has been damaged.

Elmi77, 2021-01-13:
What customer service? You sell these boards via eBay without any service or warranty!

Btw: two boards (different hardware releases) show the same problem. So this does not seem to be a unique problem of my board only but more a general error of your hardware.

lewisxhe, 2021-01-15:
As long as it is purchased in our LilyGo store, there is after-sales service. Moreover, our board has been around for a long time. It has been verified that only a few hardware is defective, but this is unpredictable.

closed 2021-03-01 by lewisxhe

vbrunko-voc, 2025-02-20:
Same problem in 2025.