Xinyuan-LilyGO/LilyGO-T-SIM7000G #74: SD Card and serialcomunication issue

stephj915, 2020-12-12:
Hi, I have discovered that when the sd card is plug in the board, I can't upload my program (header packet is missing).
I have another issue when the sd card is plugged and when i use a serial communication (via the usb cable) in same time, some stranges caracters appear during the comunication.

Thes issue is not normal, can you help me ?

Thank you in advance.

lewisxhe, 2020-12-12:
SD pin uses GPIO2, which has the function of booting into download mode, so it is impossible to download when the SD card is inserted.  This is already reminded in the README.

stephj915, 2020-12-12:
oops, sorry.
I think i am tired ...

Thank you for your reply.

closed 2020-12-12 by stephj915