Xinyuan-LilyGO/LilyGO-T-SIM7000G #75: Will this work with LTE providers in USA? And how stable is this board?

bhupiister, 2021-01-24:
I need to know if this board supports LTE/4G providers in USA?
How stable is this board?
Any power consumption test done? Deep sleep current?
Can USB type C be removed while on battery without causing board restart?
I need to purchase 10 of these boards so want to confirm before buying.

derTim, 2021-01-30:
- Can USB type C be removed while on battery without causing board restart?
Yes

Don't have any information regarding your other questions (just recieved my first board some days ago).
Had it runnning for 4 days on a single 18650 cell (after that time, still at >3.5v), waking up every 5min to connect, waiting for a gps fix and post to mqtt.
Sometimes it got no connection, but that's probably a reception/software issue on my side.

There is another closed issue with information on deep sleep current, somewhere below 1ma.

lewisxhe, 2021-03-01:
SIM7000G supports the whole world, but it does not support 4G. For 4G, you need to purchase SIM7600

closed 2021-03-01 by lewisxhe