Xinyuan-LilyGO/LilyGO-T-SIM7000G #76: LilyGO-T-SIM7000G/examples/Arduino_TinyGSM is it really working?

Paryavi, 2021-02-11:
Hi, I bought two of 7000G boards, but the TinyGSM library does not support 7000G! **so is your Arduino code really work?**
I get this error; **#error "Please define GSM modem model"**

Details that TinyGSM library does not support 7000G;
`https://github.com/vshymanskyy/TinyGSM/blob/master/src/TinyGsmClient.h`
// typedef TinyGsmSim7000::GsmClientSecureSim7000 TinyGsmClientSecure; TODO!

Thanks

procopio, 2021-02-11:
I used this example bellow few days ago, it also uses TinyGSM
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_NetworkTest/Arduino_NetworkTest.ino

lewisxhe, 2021-03-01:
Support, please refer to our example instead of running the TinyGSM built-in example directly.

closed 2021-03-01 by lewisxhe