Xinyuan-LilyGO/LilyGO-T-SIM7000G #78: capacitor parallel to R62 and R64?

alexseuf, 2021-02-15:
Hello everyone,
I have sporadically slighly wrong measurement values on ADC measurement of V_bat and Solar_in (appox. +/-5% from median value).
To reduce this, I would try to solder a SMD-100nF? Capacitor on top of R62 and R64.
Could you please tell me where I could find this resistors on the PCB?
Thank you

Mr-HaleYa, 2021-02-23:
[image]
 
If this solved the issue please mark it as closed - Hale

lewisxhe, 2021-03-01:
These two positions as shown in the figure 
[image]

closed 2021-03-01 by lewisxhe

alexseuf, 2021-03-01:
Thank you lewis he,
please confirm, that the red marked resistores in my picture are R62 and R64?
The yellow marked resistors are wrong =>100R and not 100k (picture from  Mr-HaleYa ).
Thank you
[image]

lewisxhe, 2021-03-01:
yes !

Mr-HaleYa, 2021-03-08:
@alexseuf sorry I got the wrong ones I dis say it was a guess but glad you got the right ones