Xinyuan-LilyGO/LilyGO-T-SIM7000G #81: Problem with SSL with lib "TINY_GSM_MODEM_SIM7000_CA"

Francois-37, 2021-02-26:
I use the librairy TINY_GSM_MODEM_SIM7000_CA but it didn't work and I dont know why ?
Can you help me, I spend almost 2 weeks on this problem and I didn't resolve it.
I use the exemple AllFunction and the SSL example (On SIM7000 in github)

I use many boards, only Lilygo-SIM7000G, and this program works well for some of them (I try many things and I have probably changed some features but I don't know which ones) and don't work on others.

I think what it doesn't connect to GPRS, TINY_GSM_MODEM_SIM7000_CA , but with TINY_GSM_MODEM_SIM7000 it's ok.
The problem is that I can't use SSL (https) with TINY_GSM_MODEM_SIM7000, and I need it.

I remove server and resource because it's secret and apn is good.

Thank you for help, I am really stuck.
I would be extremely grateful if you could help me clear the situation.

```
#define CAPTEUR 1
#include <Arduino.h>
//Sources :
//https://fr.aliexpress.com/item/4000542688096.html?spm=a2g0s.8937460.0.0.42ed2e0eSnVJds
//https://github.com/vshymanskyy/TinyGSM
//https://github.com/marmotton/TinyGSM/tree/SIM7000SSL/SIM7000SSL-dev (dans TinyGSM mais spécial pour le SSL)
//https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G

#define TINY_GSM_MODEM_SIM7000_CA

#define SerialMon Serial
#define SerialAT Serial1
#include <OneWire.h>          //lib I2C

#if !defined(TINY_GSM_RX_BUFFER)
#define TINY_GSM_RX_BUFFER 650
#endif

#define DUMP_AT_COMMANDS // See all AT commands, if wanted
#define TINY_GSM_DEBUG SerialMon // Define the serial console for debug prints, if needed
// #define LOGGING  // <- Logging is for the HTTP library

#define GSM_AUTOBAUD_MIN 9600  // Range to attempt to autobaud
#define GSM_AUTOBAUD_MAX 115200

// Add a reception delay - may be needed for a fast processor at a slow baud rate
// #define TINY_GSM_YIELD() { delay(2); }

// Define how you're planning to connect to the internet
#define TINY_GSM_USE_GPRS     true
#define TINY_GSM_USE_WIFI     false
#define TINY_GSM_POWERDOWN    false
#define TINY_GSM_TEST_BATTERY true

#define ORANGE                true
#define SFR                   false

#define uS_TO_S_FACTOR 1000000ULL  // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  15*60          // Time ESP32 will go to sleep (in seconds)

// set GSM PIN, if any

#define MODEM_POWER_ON 23

// flag to force SSL client authentication, if needed
// #define TINY_GSM_SSL_CLIENT_AUTHENTICATION

#define GSM_PIN ""
// Your GPRS credentials, if any
#if ORANGE
  const char apn[]  = "orange.fr";     //SET TO YOUR APN
  const char gprsUser[] = "orange";
  const char gprsPass[] = "orange";
#endif

#if SFR
  const char apn[]  = "sl2sfr";     // sl2sfr wapsfr websfr
  const char gprsUser[] = "";
  const char gprsPass[] = "";
#endif

const char server[] = "";

const int  port       = 443;
int stat_NetworkStatus = 0;
int nombre_iteration_avant_redemarrage=5;
int i=0; //en attendant la temparture
float temperature=0;
int delay_2_envois=5000;
double imei = 862649048570718;

#include <OneWire.h>
#include <DallasTemperature.h>
const int oneWireBus = 0;     
OneWire oneWire(oneWireBus);
DallasTemperature sensors(&oneWire);

#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>

// Just in case someone defined the wrong thing..
#if TINY_GSM_USE_GPRS && not defined TINY_GSM_MODEM_HAS_GPRS
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS false
#define TINY_GSM_USE_WIFI true
#endif

#if TINY_GSM_USE_WIFI && not defined TINY_GSM_MODEM_HAS_WIFI
#undef TINY_GSM_USE_GPRS
#undef TINY_GSM_USE_WIFI
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
#endif

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm        modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

TinyGsmClientSecure client(modem);
HttpClient          http(client, server, port);

void setup() {
  // Set console baud rate
  //--SerialMon.begin(115200);
  Serial.begin(9600);
  delay(10);

  // Set LED OFF
  pinMode(12, OUTPUT);
  digitalWrite(12, HIGH);

  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH);
  delay(300);
  digitalWrite(4, LOW);
  
  // !!!!!!!!!!!
  // Set your reset, enable, power pins here
  // !!!!!!!!!!!

  SerialMon.println("Wait...");
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_POWER_ON, HIGH);
  
  delay(10000);
  
  // Set GSM module baud rate
  //TinyGsmAutoBaud(SerialAT, GSM_AUTOBAUD_MIN, GSM_AUTOBAUD_MAX);
  //--SerialAT.begin(115200, SERIAL_8N1, 26, 25);
  SerialAT.begin(9600, SERIAL_8N1, 26, 27); //U_Rxd/U_Txd
  delay(6000);

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  SerialMon.println("Initializing modem...");
  if (!modem.restart()) {
    Serial.println("Failed to restart modem");
    stat_NetworkStatus=-1;
  }
  // modem.init();

  String name = modem.getModemName();
  delay(500);
  Serial.println("Modem Name: " + name);

  String modemInfo = modem.getModemInfo();
  delay(500);
  Serial.println("Modem Info: " + modemInfo);

#if TINY_GSM_USE_GPRS
  // Unlock your SIM card with a PIN if needed
  if ( GSM_PIN && modem.getSimStatus() != 3 ) {
    modem.simUnlock(GSM_PIN);
  }
#endif
  
  sensors.begin();
}

int t=0;

void loop() {

  sensors.requestTemperatures(); 
  float temperatureC = sensors.getTempCByIndex(0);
  Serial.print(temperatureC);
  Serial.println("ºC");
  delay(500);

  Serial.print("temperature : ");
  Serial.println(temperatureC);
  
  if(temperatureC == -127 || stat_NetworkStatus== -1 || stat_NetworkStatus> nombre_iteration_avant_redemarrage) {
    Serial.println("Restarting !");
    ESP.restart();
  }

  int8_t percent = batterie();
  
  char resource[300];
  sprintf(resource, "", CAPTEUR, temperatureC, percent);

#if TINY_GSM_USE_GPRS && defined TINY_GSM_MODEM_XBEE
  // The XBee must run the gprsConnect function BEFORE waiting for network!
  modem.gprsConnect(apn, gprsUser, gprsPass);
#endif

  SerialMon.print("Waiting for network...");
  if (!modem.waitForNetwork()) {
    SerialMon.println(" fail");
    delay(10000);
    return;
  }
  SerialMon.println(" success");

  if (modem.isNetworkConnected()) {
    SerialMon.println("Network connected");
  }

  // GPRS connection parameters are usually set after network registration
    SerialMon.print(F("Connecting to "));
    SerialMon.println(apn);
    //if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    if (!modem.gprsConnect(apn)) {
      SerialMon.println(" fail");
      delay(10000);
      stat_NetworkStatus++;
      return;
    }
    SerialMon.println("Connected to APN");

    delay(500);
    
    if (modem.isGprsConnected()) {
      SerialMon.println("");
      SerialMon.println("GPRS connected");
    }
    else return;

  SerialMon.print(F("Performing HTTPS GET request... "));
    http.connectionKeepAlive();  // Currently, this is needed for HTTPS
  int err = http.get(resource);
  if (err != 0) {
    SerialMon.println(F("failed to connect"));
    delay(10000);
    stat_NetworkStatus++;
    //return;
  }

  int status = http.responseStatusCode();
  SerialMon.print(F("Response status code: "));
  SerialMon.println(status);
  if (!status || status == 2) {
    delay(10000);
    stat_NetworkStatus++;
    //return;
  }

  http.stop();
  SerialMon.println(F("Server disconnected"));

  modem.gprsDisconnect();
  SerialMon.println(F("GPRS disconnected"));

  Serial.print("stat_NetworkStatus : ");
  Serial.println(status);

  if(status ==200) sleep();
}

void sleep() {
  modem.poweroff();
  Serial.println("Poweroff.");
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  delay(200);
  esp_deep_sleep_start();
}

int8_t batterie() {
  uint8_t  chargeState = -99;
  int8_t   percent     = -99;
  uint16_t milliVolts  = -9999;
  modem.getBattStats(chargeState, percent, milliVolts);
  DBG("Battery charge state:", chargeState);
  DBG("Battery charge 'percent':", percent);
  DBG("Battery voltage:", milliVolts / 1000.0F);
  return percent;
}
```

lewisxhe, 2021-02-28:
Please check SIMCom's SSL documentation.

https://www.simcom.com/technical_files.html?pro_cat=0&pro_li=10&time=0&filetype=0

Mr-HaleYa, 2021-03-30:
read the reply on this issue https://github.com/vshymanskyy/TinyGSM/issues/425#issuecomment-809795031

@lewisxhe also not a ttgo sim7000 problem that's a TinyGSM problem. so this should be closed here

KumphanartDansiri, 2021-07-03:
#define TINY_GSM_MODEM_SIM7000SSL instead of #define TINY_GSM_MODEM_SIM7000 for the 7000G model.
I am not sure about SIM7000_CA.