Xinyuan-LilyGO/LilyGO-T-SIM7000G #82: Battery Reading Is 0, solar panel voltage?

4gino, 2021-03-01:
Hi everybody, and thank you in advance.
For the latest version of the board, do you have a working example code for reading battery and solar voltage data and go to deep sleep? I tried but i keep Reading 0 for battery, i tried with 34 or 14 analog input, buy i' m confused. Thanks, gino

Mr-HaleYa, 2021-03-01:
Latest version as in Version 20200415? If so just use this 👍  (I wrote this in like 10 min but it _Should_ work )

```cpp

#define PIN_BAT       35   /* read battery voltage, Disabled when using USB - ONLY INPUT PIN */

int BP = 0;
float v_bat = 0;
const int VReads = 15;    // number of voltage readings before choosing a median, keep this number odd

void setup() {
  Serial.begin(9600);       // Start serial monitor at a baud rate of 9600
  delay(500);
  pinMode(PIN_BAT, INPUT);
}

void read_bat() {         // reads and returns the battery voltage
  float voltageBuffer[VReads];
  uint32_t Read_buffer = 0;
  for (int x = 0; x < VReads; x++) {
    for (int i = 0 ; i < VReads; i++) {
      voltageBuffer[i] = (uint32_t)analogRead(PIN_BAT);
    }
    sortData(voltageBuffer, VReads);
    Read_buffer += (voltageBuffer[(VReads - 1) / 2]);
  }
  v_bat = (((float)(Read_buffer / VReads) / 4096) * 3600 * 2) / 1000;
  BP = mapBatt(v_bat,2.5,4.2,0,100);     // get battery voltage as a percentage 0-100%
  if (BP < 0) { BP = 0; }
}

float mapBatt(float x, float in_min, float in_max, float out_min, float out_max) {
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}

Void loop(){
  read_bat();
  Serial.print("Battery Voltage : "); Serial.println(v_bat);  // 2.5v - 4.2v
  Serial.print("Battery Percentage : "); Serial.print(BP);  Serial.println("%");  // 0-100%
  delay(5000);
}
```

4gino, 2021-03-01:
Thanks, used platformio and now It works nicely, thanks again, your answer was usefull

Mr-HaleYa, 2021-03-02:
@4gino You are welcome, If your issue is fully solved please marked as Closed. 😄

closed 2021-03-02 by 4gino

carminelau, 2021-04-07:
@4gino gino can you help me ? this is my email: carminelaudato99@gmail.com and my telephone number is +39 3405073613

nravanelli, 2022-03-25:
> Latest version as in Version 20200415? If so just use this 👍 (I wrote this in like 10 min but it _Should_ work )
> 
> ```c++
> #define PIN_BAT       35   /* read battery voltage, Disabled when using USB - ONLY INPUT PIN */
> 
> int BP = 0;
> float v_bat = 0;
> const int VReads = 15;    // number of voltage readings before choosing a median, keep this number odd
> 
> void setup() {
>   Serial.begin(9600);       // Start serial monitor at a baud rate of 9600
>   delay(500);
>   pinMode(PIN_BAT, INPUT);
> }
> 
> void read_bat() {         // reads and returns the battery voltage
>   float voltageBuffer[VReads];
>   uint32_t Read_buffer = 0;
>   for (int x = 0; x < VReads; x++) {
>     for (int i = 0 ; i < VReads; i++) {
>       voltageBuffer[i] = (uint32_t)analogRead(PIN_BAT);
>     }
>     sortData(voltageBuffer, VReads);
>     Read_buffer += (voltageBuffer[(VReads - 1) / 2]);
>   }
>   v_bat = (((float)(Read_buffer / VReads) / 4096) * 3600 * 2) / 1000;
>   BP = mapBatt(v_bat,2.5,4.2,0,100);     // get battery voltage as a percentage 0-100%
>   if (BP < 0) { BP = 0; }
> }
> 
> float mapBatt(float x, float in_min, float in_max, float out_min, float out_max) {
>   return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
> }
> 
> Void loop(){
>   read_bat();
>   Serial.print("Battery Voltage : "); Serial.println(v_bat);  // 2.5v - 4.2v
>   Serial.print("Battery Percentage : "); Serial.print(BP);  Serial.println("%");  // 0-100%
>   delay(5000);
> }
> ```

@Mr-HaleYa What is your sortData function? trying to emulate this code. Thanks :)

Mr-HaleYa, 2022-07-29:
@nravanelli 
I never replied to you... 👀  so sorry... hopefully you figured it out but for future people, this is the sortData code. It's just a bubble filter

```c
void sortData(float data[], int8_t size) {             // Sorts all data small to large
  float swapper;
  for (int8_t i = 0; i < (size - 1); i++) {
    for (int8_t o = 0; o < (size - (i + 1)); o++) {
      if (data[o] > data[o + 1]) {
        swapper = data[o];
        data[o] = data[o + 1];
        data[o + 1] = swapper;
      }
    }
  }
}
```

procopio, 2023-07-25:
Sorry to came back to this topic, but I'm getting this strange reads from your function, what could be wrong please?

******* readVcc voltageBuffer = 2460.00, 2463.00, 2465.00, 2466.00, 2466.00, 2471.00, 2480.00, 2480.00, 2481.00, 2481.00, 2484.00, 2484.00, 2486.00, 2492.00, 2496.00, !
******* readVcc voltageBuffer = 2453.00, 2455.00, 2458.00, 2462.00, 2467.00, 2467.00, 2467.00, 2468.00, 2473.00, 2473.00, 2483.00, 2487.00, 2490.00, 2491.00, 2497.00, !
******* readVcc voltageBuffer = 2461.00, 2465.00, 2466.00, 2469.00, 2470.00, 2474.00, 2477.00, 2478.00, 2481.00, 2482.00, 2483.00, 2485.00, 2487.00, 2496.00, 2496.00, !
******* readVcc voltageBuffer = 2461.00, 2461.00, 2463.00, 2465.00, 2467.00, 2471.00, 2474.00, 2479.00, 2480.00, 2481.00, 2482.00, 2485.00, 2486.00, 2487.00, 2496.00, !
******* readVcc voltageBuffer = 2466.00, 2474.00, 2475.00, 2479.00, 2480.00, 2482.00, 2482.00, 2485.00, 2486.00, 2488.00, 2489.00, 2494.00, 2494.00, 2495.00, 2495.00, !
******* readVcc voltageBuffer = 2457.00, 2465.00, 2475.00, 2475.00, 2475.00, 2478.00, 2480.00, 2480.00, 2480.00, 2480.00, 2481.00, 2481.00, 2481.00, 2481.00, 2495.00, !
******* readVcc voltageBuffer = 2461.00, 2465.00, 2468.00, 2470.00, 2471.00, 2474.00, 2475.00, 2480.00, 2480.00, 2484.00, 2485.00, 2488.00, 2490.00, 2493.00, 2493.00, !
******* readVcc voltageBuffer = 2459.00, 2465.00, 2467.00, 2467.00, 2469.00, 2475.00, 2475.00, 2477.00, 2478.00, 2480.00, 2482.00, 2487.00, 2493.00, 2494.00, 2496.00, !
******* readVcc voltageBuffer = 2457.00, 2462.00, 2467.00, 2467.00, 2470.00, 2471.00, 2474.00, 2476.00, 2477.00, 2481.00, 2489.00, 2492.00, 2493.00, 2495.00, 2498.00, !
******* readVcc voltageBuffer = 2455.00, 2459.00, 2459.00, 2461.00, 2469.00, 2470.00, 2471.00, 2474.00, 2478.00, 2480.00, 2480.00, 2490.00, 2491.00, 2498.00, 2499.00, !
******* readVcc voltageBuffer = 2455.00, 2462.00, 2464.00, 2471.00, 2471.00, 2471.00, 2475.00, 2476.00, 2479.00, 2479.00, 2482.00, 2486.00, 2486.00, 2489.00, 2491.00, !
******* readVcc voltageBuffer = 2462.00, 2465.00, 2469.00, 2469.00, 2469.00, 2469.00, 2475.00, 2476.00, 2479.00, 2479.00, 2480.00, 2481.00, 2483.00, 2492.00, 2494.00, !
******* readVcc voltageBuffer = 2465.00, 2465.00, 2466.00, 2469.00, 2469.00, 2470.00, 2474.00, 2475.00, 2475.00, 2475.00, 2479.00, 2480.00, 2483.00, 2485.00, 2491.00, !
******* readVcc voltageBuffer = 2464.00, 2467.00, 2469.00, 2469.00, 2471.00, 2471.00, 2474.00, 2476.00, 2478.00, 2479.00, 2479.00, 2480.00, 2481.00, 2481.00, 2488.00, !
******* readVcc voltageBuffer = 2466.00, 2469.00, 2471.00, 2474.00, 2477.00, 2480.00, 2480.00, 2481.00, 2483.00, 2486.00, 2486.00, 2487.00, 2495.00, 2496.00, 2496.00, !
******* readVcc Read_buffer, bat, BP = 37161 , 4.35 , 109.06
batteryVoltage = 109%

```
void sortData(float data[], int8_t size) {             // Sorts all data small to large
    float swapper;
    for (int8_t i = 0; i < (size - 1); i++) {
        for (int8_t o = 0; o < (size - (i + 1)); o++) {
            if (data[o] > data[o + 1]) {
                swapper = data[o];
                data[o] = data[o + 1];
                data[o + 1] = swapper;
            }
        }
    }
}

float mapBatt(float x, float in_min, float in_max, float out_min, float out_max) {
    float ret = (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
    if (ret < 0) { ret = 0; }
    return ret;
}

int readVcc() {         // reads and returns the battery voltage
    int VReads = 15;
    float voltageBuffer[VReads];
    uint32_t Read_buffer = 0;
    for (int x = 0; x < VReads; x++) {
        for (int i = 0; i < VReads; i++) {
            voltageBuffer[i] = (uint32_t)analogRead(BATTERY_POWER_PIN);
        }
        sortData(voltageBuffer, VReads);

        DEBUG_print(F("******* readVcc voltageBuffer = "));
        for (size_t i = 0; i < VReads; i++) {
            DEBUG_print(voltageBuffer[i]);
            DEBUG_print(F(", "));
        }
        DEBUG_println(F("!"));

        Read_buffer += (voltageBuffer[(VReads - 1) / 2]);        
    }

    float bat = (((float)(Read_buffer / VReads) / 4096) * 3600 * 2) / 1000;
    float BP = mapBatt(bat, 2.5, 4.2, 0, 100);     // get battery voltage as a percentage 0-100%

    DEBUG_print(F("******* readVcc Read_buffer, bat, BP = "));
    DEBUG_print((long)Read_buffer);
    DEBUG_print(F(" , "));
    DEBUG_print(bat);
    DEBUG_print(F(" , "));
    DEBUG_println(BP);

    return (int)BP;
}
```